UBC-Thunderbots/Software #594: Make PathPlanningNavigator

jonl112, 2019-05-04:
### Description of the task
Replace PlaceholderNavigator in ai.cpp with your new navigator that, for now, calls StraightLinePathPlanner. Assigned primitives should be the first primitive in the path for each of the robots.

### Outline the task in more detail
Depends on #357 
Blocks field testing of navigator

### Acceptance criteria

- [ ] Make new navigator 
- [ ] Change ai.cpp to call your navigator 
- [ ] Unit testing/field testing

akhilveeraghanta, 2019-05-20:
Did some preliminary setup work to call path planning navigator and setting up the file structure. And handed it off to @jonl112  as he would like to work on passing in obstacles and finish up the rest.  We need a new ticket to work on all the other intents that may need navigator once the path planner and the basic navigator is merged in

MathewMacDougall, 2019-05-22:
Currently only the MoveIntent uses navigation. We can make any new tickets once the navigator is in.

closed 2019-07-02 by akhilveeraghanta