UBC-Thunderbots/Software #626: Decrease the zoom step in the visualizer

MathewMacDougall, 2019-06-07:
### Description of the task

Right now each "step" of zoom in the visualizer is relatively large, and makes it more difficult to fit the view of the field in the available window space without too much or too little overlap.

The size of the step for each zoom level should be reduces accordingly.

### Outline the task in more detail

Reduce the size of the zoom step when zooming in the field view, ie. slow down the zooming.

### Acceptance criteria

- [ ] Slow down the zoom in the field view

qgolsteyn, 2019-06-07:
This is a Firefox bug which has been hard to figure out. Essentially, it sends larger scroll deltas compared to Chrome, which then results in larger jumps in the zoom on the visualizer. I will continue to investigate.

qgolsteyn, 2019-06-07:
Probably a bug with `pixi-viewport`, the library we are using to control the camera. Might try to push a PR there to fix the issue at the source.

MathewMacDougall, 2019-06-07:
Oh if it's browser specific then don't bother for now. We have higher-value work to finish first

closed 2019-10-06 by MathewMacDougall