UBC-Thunderbots/Software #634: Ball Filter improvements

MathewMacDougall, 2019-06-12:
Currently a list of ideas to improve the ball filter

Problems:
* Does not handle slow balls very well. Noise has an extra detrimental effect, and doesn't maintain good accuracy of velocity magnitude and direction.
* Even at faster ball speeds, "moderate" noise can have a more severe impact than expected. Single ticks with angles deviating by up to 8 degrees is not good.

Ideas / Solutions:
* Use the portion of the ball detection buffer proportional to the ball's speed. The higher the speed the smaller the buffer
* For velocity estimation, try all combinations of a detection with points in the future and then take average, instead of just point to point. Hopefully will give better average. Should the points be projected onto the line of best fit before estimating velocities? using the "correct" positions might help with accuracy too.

garethellis0, 2019-11-20:
We really should have a more formal approach to this, likely around some model of the ball's dynamics and a noise model, perhaps something like a EKF?

MathewMacDougall, 2019-11-20:
We should have an in-person discussion. I do like the simplicity of the regression.

garethellis0, 2019-11-21:
Right, but I think if we want any significant improvement beyond this we need to go for something more sophisticated. I think we're probably nearing the upper bound for what we can reasonably achieve with regression, and I'd much rather we spend time doing the "more formal" way then adding tweaks/hacks to improve regression.

MathewMacDougall, 2019-11-21:
Fair enough, lets meet in person when we're ready to start tackling this.

akhilveeraghanta, 2020-09-06:
Closing for lack of activity and too broad of a scope

closed 2020-09-06 by akhilveeraghanta

MathewMacDougall, 2020-09-08:
This is actually basically done in the current filter anyway.