UBC-Thunderbots/Software #678: Make point == operator use EPS2 to check for equality

jonl112, 2019-07-02:
### Description of the task
We do a lot of (p1-p2).lensq()<EPS2. We don't want to do that everywhere. We want to be able to just use == instead.

### Outline the task in more detail
This caused bugs in the past. This will avoid future bugs

### Acceptance criteria

- [ ] Make point == operator use EPS2 to check for equality
- [ ] Refactor all code to use ==
- [ ] Testing

MathewMacDougall, 2019-07-24:
Where do we do these kind of comparisons? The current `==` operator for the `Point` class already uses an EPS value as suggested. Are the bugs from the past reproducible? We would probably want to consider making the EPS value slightly larger in that case, since right now it's the absolute smallest it can reasonably be.

jonl112, 2019-07-24:
Mostly in geom/util.cpp. Maybe the task is to evaluate switching them all to ==

MathewMacDougall, 2019-07-31:
Ah I see what you mean. That makes more sense to me then.

What bugs did this cause in the past?

garethellis0, 2019-11-20:
@jsn0716 @scveloso - please make sure to account for this in the geom re-write, feel free to close if no longer relevant.

jonathanlew, 2020-04-17:
I think that new geom has addressed this with FIXED_EPSILON, unit testing, etc. so we can close it.

closed 2020-04-17 by jonathanlew