UBC-Thunderbots/Software #735: Create BallState class and use this to store ball state

MathewMacDougall, 2019-08-05:
### Description of the task
Create a `BallState` class to store all existing ball state at a single point in time, and have the `Ball` class store a buffer of `BallState` in order to preserver history.

### Outline the task in more detail

Currently, we store the history of the ball state by maintaining a buffer of previous positions, velocities, timestamps, etc. Maintaining a buffer for each member variable is cumbersome, and opens us up for mistakes since we don't have guarantees that the buffers will all be the same length. (So for example you can't loop over one buffer and access data in another buffer at the same index, because one might be longer than the other).

Therefore, we think it's better to separate the "state" of the ball, and the class that stores state.

A new `BallState` class should be created, that represents a ball's state at a single point in time / aka a single timestamp. This new class will not have any buffers, but will still have the same member variables like `position`, 'velocity', 'timestamp'.

The current `Ball` class should then store a single buffer of `BallStates` in order to store history. The `Ball` class should still expose the same interface for getting the current data. Eg. `ball.position()`, `ball.velocity()` should still exist.

The only function related to the ball's "state history" should be "getBallStateHistory" which returns the values in the buffer.

There should be very few places, if any, that currently use the Ball's `getPreviousPositions()` functions. This should make changing the classes relatively safe since they expose the same interface.

Tests will need to be updated. Most existing `Ball` tests would just become `BallState` tests now. The new `Ball` class would need tests to make sure it returns the correct `position`, `velocity` values (ie. making sure we're returning from the right end of the buffer), and that the history it returns is expected.

### Acceptance criteria

- [ ] ICreate a new `BallState` class
- [ ] Implement the `BallState` class
- [ ] Update unit tests for the `BallState` class
- [ ] Update the `Ball` class to use `BallState`
- [ ] Update unit tests for the new `Ball` class

garethellis0, 2019-11-20:
@jared-paul - this is finished right? Please close if so, or leave a comment with remaining work to do if there is any.

garethellis0, 2020-01-06:
@jared-paul ?

jared-paul, 2020-01-06:
Yep this is finished, sorry about that

closed 2020-01-06 by jared-paul