UBC-Thunderbots/Software #738: parameter object initialization order issue

akhilveeraghanta, 2019-08-15:
### Description of the task
Provide a way to initialize parameters, instead of relying on static initialization.

### Outline the task in more detail
Since we just use namespaces and don't have a singleton or class structure, and because c++ doesn't guarantee any order of initialization, there is no guarantee parameters will be ready when other constructors/objects need them at the very beginning of the program.

So in the functions `value` and `setValue`, we should only be returning the value of the parameter in the registry, not the value of the parameter instance used to access the registry. (We currently do so, so if any objects are initialized before the parameters are initialized, we return T() as there is nothing in the registry during this time.)

### Acceptance criteria

- [ ] Figure out a way and fix initialization order to make sure parameters get initialized first
- [ ] Remove return this->value and throw exception instead
- [ ] Unit tests

garethellis0, 2019-11-20:
@akhilveeraghanta - is this still relevant? Please close if not, or place in the context of the new parameter system if so.

akhilveeraghanta, 2019-11-20:
This is no longer an issue, we can close it

closed 2019-11-20 by akhilveeraghanta