UBC-Thunderbots/Software #741: Finish up Pivot Action and Primitive

henryymliu, 2019-08-20:
### Description of the task

@akhilveeraghanta repurposed the pivot primitive to have the robot pivot with the ball in the dribbler, similar to what CMU has done for years (and Zjunlict in this year's comp). Work has been started during competition but it hasn't been finished; IIRC some tuning in both firmware and HL is needed to make it work.

### Outline the task in more detail

See #731 for the diff. 
In firmware:
- `radius` can be adjusted (distance between ball centre and robot centre)
- How the elements of `accel` are calculated might also need to be tested

In HL:
- The dynamic parameters `arb_scaling` and `finish_angle_threshold` can be tuned during runtime.

### Acceptance criteria

- [ ] Pivot like CMU

garethellis0, 2019-11-20:
What goal does this fall under this year (see `G# - ` labels). If we can't justify it for at least one specific goal we should really just close this.

MathewMacDougall, 2019-11-20:
Also given the state of new firmware / old firmware cleanup it might be best to close this and re-address is when we finish cleanup, and can re-define what primitives we want with a new radio protocol

akhilveeraghanta, 2019-11-20:
Yeah agreed, and I think in its current form it doesn't do what we want it to, pivoting around the ball is handled by the navigator, this primitive will need to be redefined and rewritten

closed 2019-11-20 by MathewMacDougall