UBC-Thunderbots/Software #745: Perform proper goalie assignment using Plays

MathewMacDougall, 2019-09-06:
### Description of the task

Have the goalie specified explicitly in Plays and assigned directly, rather than relying on the cost functions.

### Outline the task in more detail

Currently we are hacking around our convention of tactic robot cost when assigning goalies. The goalie tactic applies a very high cost (eg. `1000000`) for non-goalie robots and a cost of 0 for the goalie robot. This currently defies our convention of keeping costs in the range `[0,1]`. We should rework the Plays to explicitly assign the goalie tactic, and have the tactic assignment handle this directly rather than using the cost system.

### Acceptance criteria

- [ ] Figure out how to return an explcit "goalie tactic" in Plays
- [ ] Have tactic assignment assign the goalie tactic directly
- [ ] Update all plays to use the new system

garethellis0, 2019-11-20:
Marking as difficulty 13 because of the likely required architecture changes.

garethellis0, 2019-12-31:
Resolved via #1108

closed 2019-12-31 by garethellis0