UBC-Thunderbots/Software #761: Open Pull Request to ssl_vision To Document Shared Protos

garethellis0, 2019-09-20:
### Description of the task
There are a shared set of protobuf files that the `ssl_vision` system uses to communicate with all the teams. These are found in the `ssl_vision` repo [here](https://github.com/RoboCup-SSL/ssl-vision/tree/master/src/shared/proto). They are not documented. We have copied and documented them in `src/thunderbots/software/proto`. Open a pull request to the `ssl_vision` repo with the documented proto files.

Similarly for the grsim and ssl-controller repos.

### Acceptance criteria

- [ ] Read through our Proto messages and ensure the documentation is accurate / sane (this will probably involve asking @MathewMacDougall / @henryymliu  lots of questions)
- [ ] Open a PR to the [grsim repo](https://github.com/RoboCup-SSL/grSim) to document all the relevant proto files
- [ ] Open a PR to the [ssl-vision repo](https://github.com/RoboCup-SSL/ssl-vision) to document all the relevant proto files
- [ ] Open a PR to the [ssl-game-controller repo](https://github.com/RoboCup-SSL/ssl-game-controller) to document all the relevant proto files

### Blocked By

jonathanlew, 2020-07-30:
The documentation seems ok overall; any specific areas of concern?

MathewMacDougall, 2020-07-31:
IIRC this was mostly in the context of the field lines / arcs in the geometry message, because it was not stated what all the possible string values were. This has been _partially_ resolved by their introduction of some optional enums, but doesn't solve the issue completely.

@garethellis0 was there anything else, and how strongly do you feel about this? I think a better option here may be to open an issue (or PR if you're big brain) asking them to replace the geometry strings with the enum entirely because that solves the core issue.

garethellis0, 2020-08-08:
IMO this is so low priority it's not worth our time, but yes, would be nice to remove strings wherever possible.

jonathanlew, 2020-08-08:
I don't think they can remove the strings for backward compatibility, so I'd be in favour of closing this issue

MathewMacDougall, 2020-08-10:
fine with me

closed 2020-08-16 by jonathanlew