UBC-Thunderbots/Software #763: Seperate "FreeKickPlay" into "DirectFreeKickPlay" and "IndirectFreeKickPlay"

garethellis0, 2019-09-20:
### Description of the task
- Separate direct and indirect free kick plays (probably have a generic `FreeKickPlay` that they both extend)
- Delete `CornerKickPlay` and use `IndirectFreeKickPlay` instead.
  - This will require changing the cherry-pick region assignment to be more based on ball location
- `DirectFreeKickPlay` is going to be basically the same as `IndirectFreeKickPlay` except that it can check for direct shots.

### Acceptance criteria

- [ ] Make `IndirectFreeKickPlay`
- [ ] Make `DirectFreeKickPlay`
- [ ] Remove `CornerKickPlay` and use `IndirectFreeKickPlay` instead

### Blocked By

matthewberends, 2019-11-21:
It has been decided that at robocup this year we only want to have a `DirectFreeKickPlay` as opposed to a separate `IndirectFreeKickPlay`.

### Updated Description of the task
- Remove `IndirectFreeKickPlay` and replace it with a `DirectFreeKickPlay`
- `DirectFreeKickPlay` is going to be basically the same as the current`IndirectFreeKickPlay` except that it can check for direct shots.
- Delete `CornerKickPlay` and use `IndirectFreeKickPlay` instead.

### Acceptance criteria

- [ ] Remove `IndirectFreeKickPlay`
- [ ] Make `DirectFreeKickPlay`

### Blocked By

garethellis0, 2019-11-22:
Sounds good, feel free to just directly update the issue description in the future.

garethellis0, 2020-01-06:
@matthewberends - was this completed a while back?

matthewberends, 2020-01-07:
> @matthewberends - was this completed a while back?

No this hasn't been done. This slipped under my radar over the break. Ill try to get it done this week

PR #1127 (2020-01-23): Merge indirect free kick

## Please fill out the following before requesting review on this PR

### Description
Remove the indirect free kick play and merge it into a direct free kick play. The direct free kick play will be similar to the previous indirect free kick play except that it also checks for an open shot on goal in addition to a pass and chip.

### Resolved Issues

resolves #763 

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function definitions (usually in the `.h` file) sh [... cut]
closed 2020-01-23 by matthewberends