UBC-Thunderbots/Software #764: Design a "Sensor Fusion" Layer To Seperate Network Input And Sensor Fusion

garethellis0, 2019-09-20:
### Description of the task
- We should logically separate reading raw vision data from filtering
- Filtering should also read radio breakbeam data (more generally any data from the robots) to help know where the ball is
- Removing expired ball and team data should be done regardless of cameras being enabled / zones being disabled, so we remove things properly and keep updating if we disable a camera. 
- General idea here is we need to have a seperate layer to integrate data from both the robots and network.

This ticket is for planning work and making a bunch of smaller tickets to implement that work.

### Acceptance criteria

- [ ] Create all tickets necessary to implement a "Sensor Fusion" layer, making sure to reference this ticket in all of them

### Blocked By

MathewMacDougall, 2019-11-16:
[image]

jonathanlew, 2019-11-16:
:falkland_islands: :

MathewMacDougall, 2019-11-24:
After discussion with @jonathanlew this issue is blocked by #1065 . We will merge whatever has currently been done  that's useful (folder structure, etc) but otherwise this issue will be on hold for now as we focus on higher-priorities like mrftest (and work on unblocking this issue).

PR #1134 (2020-02-06): Sensor fusion Design

### Description
- New network client to produce for new backend for sensor fusion
- New filters to support new backend
- New and updated data structures to support network client
- Update sensor fusion for new architecture with filters

### Testing Done
TBD

### Resolved Issues
resolves #764, resolves #1070

### Length Justification

A lot of is copied from backend/input/network/filter and backend/input/network/networking and is marked 
### Review Checklist
closed 2020-02-06 by jonathanlew