UBC-Thunderbots/Software #766: Redesign PassGenerator To Be Deterministic

garethellis0, 2019-09-20:
### Description of the task
Currently, `PassGenerator` is not deterministic because the optimization it uses occurs in a seperate thread. This is by-design so that it is performant in game scenarios, but it means that tests for it are quite flaky. We should redesign it to separate out the threading if possible (or at least add a compiler flag to disable it) and add a way to run it for a set number of iterations for the testing purposes.

### Acceptance criteria

- [ ] Seperate out threading from the `PassGenerator` class entirely, or failing that, add some sort of compile-time option to disable it
- [ ] Re-enable all `PassGenerator` tests and ensure that they pass.
- [ ] Consider: what about for other tests for plays that use the PassGenerator? We need to make those deterministic as well. This likely means we're missing some acceptance criteria here.....

### Blocked By

jonathanlew, 2020-01-25:
@garethellis0 is this done?

jonathanlew, 2020-05-09:
> Gareth Ellis :penguin:  12:29 PM
Partly done. It's deterministic, but you need to pass a bool into the constructor, we needed the dynamic param changes to allow us to share some "deterministic" param between all the play configs that use the passgenerator

garethellis0, 2020-05-09:
Thanks for following up on this @jonathanlew !

PR #1980 (2021-04-08): Zone based Passing

## Please fill out the following before requesting review on this PR

### Description
Initial doc here: https://docs.google.com/document/d/1ag04aEhdKtKG3IhS16Tw_tSx1Q3B2Bq7ER7e4KIh_g4/edit
Arch doc here: https://docs.google.com/document/d/14clpTMeEFrAMnvzEfV9XoediuuFyDeWB4ktByog2EKQ/edit

The zone based passing system is mainly to share computation between passers and cherry pickers and make writing game play easier.
### :spiral_notepad: NOTE to reviewers, among all files, please specifically focus on:
- `zone_pitch_division.h`
- `eighteen_zone_pitch_division.h/cpp`
- `pass [... cut]
closed 2021-04-08 by akhilveeraghanta