UBC-Thunderbots/Software #767: Review All Usages Of "loop_forever" in Tactics/Actions

garethellis0, 2019-09-20:
### Description of the task
Review all usages of the `loop_forever` argument in actions and tactics and check if they are sane. Remove any usages that don't make sense, and in cases where we remove a usage, we should assign a reasonable default in the class constructor instead.

Consider:
- should the tactic/action even have the option to `loop_forever`? Does that make sense? (ex. should `goalie` have the option to `loop_forever` or is that just implied?)

### Acceptance criteria

- [ ] Review the usage of `loop_forever` in all tactics/actions
- [ ] For all tactics/actions, if "loop_forever" is exposed and it should not be (ex. in the case of the goalie), it should not be accessible from outside the class and should be assigned a reasonable default in the class constructor.
### Blocked By

HaomiaoZ, 2019-10-08:
So I went over all the actions and tactics and noticed the following:

Actions:

-  All the `Action` are not `loop_forever` by either setting the `loop_forever` to `false` as default in the constructor or passing `false` in the unit tests when trying to construct the `Action`. 
- Currently, I cannot think of a situation where an `Action` needs to `loop_forever`.

Tactics:

- `cherry_pick`, `crease_defender`, `goalie`, and `patrol` are having `loop_forever`. All those four tactics make sense to me to have `loop_forever` as default. 
-  The robot needs to be constantly moving to the best spot based on the situation for receiving so `cherry_pick` needs to be `loop_forever`.  
- When defending, the defenders need to be constantly moving to try to block the best shot so `crease_defender` makes sense to me. 
- The goalie needs to be constantly trying to block the shot so `goalie` should be `loop_forever`.
- `patrol` tactic needs the robot to continuously move around a point so the tactic needs to be `loop_forever` that the robot can continuously move when `patrol` is assigned.
- For the rest of the `Tactics`, it makes sense to me that they are not repeating themselves over and over again.

Overall everything makes sense to me. What do you guys think? @garethellis0 @MathewMacDougall

garethellis0, 2019-10-08:
@HaomiaoZ - so are you saying we can remove the `loop_forever` parameter from all the tactics / actions?

HaomiaoZ, 2019-10-12:
@garethellis0  Sorry for the late reply, I think we still need to keep `loop_forever` parameter to indicate whether a specific tactic needs to loop forever. For example, `cherry_pick`,`crease_defender` and `patrol` still need `loop_forever` to be `true` to loop forever, while the rest of the tactics need to specify`loop_forever` as `false` to indicate that the tactic does not need to loop forever.

overall, I think we can keep all the `loop_forever` as they are now.

MathewMacDougall, 2019-10-12:
And just to double check, none of the Tactics use any Actions with `loop_forever = true`? If that's the case it seems like `loop_forever` could be removed from the Actions entirely.

@HaomiaoZ what about the ShadowEnemyTactic? Seems like it could always be loop_forever based on the same argument for crease defender or cherry-pick.

garethellis0, 2019-10-13:
Discussed this with @HaomiaoZ today, realized I wasn't quite clear enough. What I'm really asking here is where we can remove `loop_forever` from the external interfaces of a class, we can still have some tactics loop forever, we just won't be able to control if they do or not.

@HaomiaoZ - feel free to correct me if I'm wrong on this, but you thought we could remove `loop_forever` from the actions, but still had to check for usages of it for tactics in plays?

HaomiaoZ, 2019-10-19:
@garethellis0 I double-checked all the tactics again and found that some actions are set with `true` for `loop forever` in tactics. Below are the current action usages in the tactics regarding `loop forever`:

1. Block Shot Path Tactic:
    -`Move Action` is set to `false`
2. Cherry-pick Tactic:
   -`Move Action` is set to `true`
3. Chip Tactic:
    -`Chip Action` is called without arguments. `Chip Action` does not have a `loop_forever` member
4. Crease Defender Tactic:
    -`Move Action` is set to `false`
    -`Stop Action` is using the default value of `loop_forever`: `false`
5. Goalie Tactic:
    -`Move Action` is using the default value of `loop_forever`: `false`
    -`Chip Action` is called without arguments. `Chip Action` does not have a `loop_forever` member
    -`Stop Action` is using the default value of `loop_forever`: `false`
6. Move Tactic:
    -`Move Action` is set to `false`
7. Pass Tactic:
    -`Move Action` is set to `true`
    -`Kick Action` is called without arguments. `Kick Action` does not have a `loop_forever` member
8. Patrol Tactic:
    -`Stop Action` is set to `true`
    -`Move Action` is set to `false`
9. Penalty Kick Tactic:
    -2 `Move Action` are set to `false`
    -`Kick Action` is called without arguments. `Kick Action` does not have a `loop_forever` member
10. Receiver Tactic:
      -`Move Action` is set to `true`
11. Shadow Enemy Tactic:
      -`Move Action` is set to `false`
      -`Stop Action` is set to `true`
12. Shadow Freekicker Tactic:
      -`Move Action` is using the default value of `loop_forever`: `false`
13. Shoot Goal Tactic:
      -`Chip Action` is called without arguments. `Chip Action` does not have a `loop_forever` member
      -`Kick Action` is called without arguments. `Kick Action` does not have a `loop_forever` member
      -`Move Action` is set to `true`
14. Stop Tactic:
      -`Stop Action` is set to `false`

Overall, It seems that only `Stop Action` and `Move Action` need the user to set the status of `loop forever`.

Also, it seems that there are several actions that are not being used in tactics: `Dribble`, `Movespin`, and `Pivot`.

Please let me know what do you think.

HaomiaoZ, 2019-10-19:
I will leave a comment later for all the tactics usages in plays

MathewMacDougall, 2019-10-20:
Nice summary.

Does it even make sense for the `StopAction` to have the option to loop forever? I can't think of a use case where we actually wait to see if we are "done stopping", so maybe we can remove that option entirely.

I'm also slightly tempted to not give `loop_forever` a default value with the `MoveAction`. This summary shows its used with both `true` and `false` so there's not really a "reasonable default assumption", and forcing the value to be specified might save us from issues in the future of forgetting there's that behavior there. Thoughts?

garethellis0, 2019-10-20:
I think:
- `StopAction` shouldn't have an option to loop forever, should always loop
- `MoveAction` should have the option but should not have a default (for the reasons Mathew stated)
- As for the actions not being used in this tactic, that's out-of-scope for this issue, but maybe we should file an issue to review them and remove the ones we don't use/need?

MathewMacDougall, 2019-10-20:
> As for the actions not being used in this tactic, that's out-of-scope for this issue, but maybe we should file an issue to review them and remove the ones we don't use/need?

Maybe we should start a superticket of STP cleanup and changes? It's kinda tied to firmware but we either want to remove them or update them to be actually useful. But that generally means we need to think about strategy and be able to test.

garethellis0, 2019-10-20:
Yeah, if we had a list of things to review in Jan/Feb or so that would be good (a ticket would be a good place for this), hopefully by then it will be obvious if we can nuke these actions or not.

HaomiaoZ, 2019-10-20:
I agree on `Move Action` should not have a default value as well since there are cases for using `true` and `false`. For `Stop Action`, I currently have no thought since I am not sure how the `Stop Action` is used with default `loop forever` value in `Crease Defender` and  `Goalie`

MathewMacDougall, 2019-10-20:
Made #968 to track general STP stuff for now

garethellis0, 2019-10-20:
Ok, so action items here are:
- remove all instances of `loop_forever` except for `StopAction` and `MoveAction` (see above comments for those) in both actions and tactics

@HaomiaoZ @MathewMacDougall ?

MathewMacDougall, 2019-10-20:
I say remove `loop_forever` in the `StopAction` as well.

HaomiaoZ, 2019-10-24:
Here is the summary of the tactic usage in the plays:

1. Corner Kick Play:
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - `Move Tactic`: 1 `align_to_ball_tactic` is using the defualt value of `loop_forever`: `false`.2 `bait_move_tactic` set as `true`
     - 2 `Cherry Pick Tactic` are using the default value of `loop_forever`: `true`
     - 1`Passer Tactic` set `loop_froever` as `false`
     - 1`Receiver Tactic` set `loop_forever` as `false`

2. Defense play:
    - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 3 `Shadow Enemy Tactic` set `loop_forever` as `true`
     - 2 `Crease Defender Tactic` are using the default value of `loop_forever`: `true`
     - 3 `Stop Tactic` set `loop_forever` as `true` for fallback

3. Enemy Free kick play:
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 1 `Crease Defender Tactic` is using the default value of `loop_forever`: `true`
     - 2 `Shadow Freekicker Tactic` set `loop_forever` as `true`
     - 3 `Shadow Enemy Tactic` set `loop_forever` as `true`
     - 2 `Move Tactic` set as `true`

4. Halt play:
     - 6 `Stop Tactic` set as `true`

5. kickoff enemy play:
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 3 `Shadow Enemy Tactic` set `loop_forever` as `true`
     - 5 `Move Tactic` set as `true`

6. Kickoff Friendly play:
     - 5 `Move Tactic` set as `true`
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 1 `Chip Tactic` set as `true`

7. Penalty Enemy:
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 5 `Move Tactic` set as `true`

8. Penalty kick play:
     - 1 `Panealty Kick Tactic` set as `true`
     - 5 `Move Tactic` set as `true`

9. Shoot or chip play:
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 2 `Crease Defender Tactic` are using the default value of `loop_forever`: `true`
     - 2 `Move Tactic` set as `true`
     - 1 `Shoot Goal Tactic` set to `false`

10. Shoot or Pass Play
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
     - 2 `Crease Defender Tactic` are using the default value of `loop_forever`: `true`
     - 2 `Cherry Pick Tactic` are using the default value of `loop_forever`: `true`
     - 1 `Shoot Goal Tactic` set to `false`
      - 1`Passer Tactic` set `loop_froever` as `false`
     - 1`Receiver Tactic` set `loop_forever` as `false`

11. Stop Play
     - 5 `Move Tactic` set as `true`
     - 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`

HaomiaoZ, 2019-10-24:
Agree on removing `loop_forever` for all `Action` other than `Stop Action` and `Move Action` . Still no opinion on `Stop Action` since I haven't really take a look of the logic in the code yet for `Stop Action` usage in `Tactic`

MathewMacDougall, 2019-10-26:
> Still no opinion on Stop Action since I haven't really take a look of the logic in the code yet for Stop Action usage in Tactic

I'd say remove the `loop_forever` parameter but leave the other condition in the loop where it checks for robot speed.

As for the tactics, here's my thoughts based on the summary:
**Tactics whose `loop_forever` parameter should be removed**
* GoalieTactic should not expose the `loop_forever` parameter and should default to true
* CherryPickTactic should not expose the `loop_forever` parameter and should default to true
* ShadowEnemyTactic should not expose the `loop_forever` parameter and should default to true
* CreaseDefenderTactic should not expose the `loop_forever` parameter and should default to true
* StopTactic should not expose the `loop_forever` parameter and should default to true
* ShadowFreeKickerTactic should not expose the `loop_forever` parameter and should default to true

**Tactics that should keep the `loop_forever` and have the default value removed**
* MoveTactic should keep the `loop_forever` parameter and the default value should be removed
* PasserTactic should keep the `loop_forever` parameter and the default value should be removed (@garethellis0 would we ever want to loop forever with a passer?)
* ReceiverTactic should keep the `loop_forever` parameter and the default value should be removed (@garethellis0 would we ever want to loop forever with a receiver?)
* ChipTactic should keep the `loop_forever` parameter and the default value should be removed
* PenaltyKickTactic should keep the `loop_forever` parameter and the default value should be removed
* ShootGoalTactic should keep the `loop_forever` parameter and the default value should be removed

MathewMacDougall, 2019-10-26:
Oh @HaomiaoZ once #972 is merged (probably early Saturday morning) can you take a look at the tactics being used again? There will some extra ones being used like the `PatrolTactic` and stuff and we don't want to forget those.

I think the `BlockShotPath` tactic won't be used at all so it could be marked for removal, but lets confirm once that PR is merged (that's the last of the RoboCup changes). Sorry I totally forgot all the changes I've been making this week would affect you.

HaomiaoZ, 2019-11-09:
I went over all the tactics and plays again, and here are the updates:

**New tactics:**

Defense Shadow Enemy Tactic:
- `Move Action` is using the default value of `loop_forever`: `false`
- `Stop Action` is set to `true`

Grab ball Tactic:
- `Intercept Ball Action` is set to `true`
-  `Move Action` is using the default value of `loop_forever`: `false`
- `Movespin Action` is called without arguments. `Movespin Action` does not have a `loop_forever` member

Kickoff Chip Tactic:
- Inherited from `Chip Tactic`. `Chip Tactic` constructor is called with a potential argument for `loop_forever`. The default `loop_forever` for `Kickoff Chip Tactic` is set to `false`

Penalty Setup Tactic:
- Inherited from Move Tactic. `MoveTactic` constructor is called with a potential argument for `loop_forever`. The default `loop_forever` of `Penalty Setup Tactic` is set to `false`

**New plays:**

Indirect Free Kick Play:
- 1 `Goalie Tactic` is using the default value of `loop_forever`:`true`
- 2 `Crease Defender Tactic` are using the default value of `loop_forever`: `true` 
- 2 `Cherry Pick Tactic` are using the default value of `loop_forever`: `true`
- 1 `Move Tactic` is using the defualt value of `loop_forever`: `false`
- 1 `Chip Tactic` is using the defualt value of `loop_forever`: `false`
- 1 `Passer Tactic` set `loop_froever` as `false`
- 1 `Receiver Tactic` set `loop_forever` as `false`

**Updated plays:**

Below are the current usages of tactics in modified plays:

Defense play:
- 1 `Goalie Tactic` is using the default value of `loop_forever`: `true` 
- 1 `Grab ball Tactic` set  `loop_forever` to `true` (new)
- 1 `Shoot Goal Tactic` set  `loop_forever` to `false` (new)
- 1 `Defense Shadow Enemy Tactic` is using the default value of `loop_forever`: true (new)
- 1 `Shadow Enemy Tactic` is using the default value of `loop_forever`: `false` (Changed from 3  to 1)
- 2 `Crease Defender Tactic` are using the default value of `loop_forever`: `true`
- 2 `Move Tactics` set  `loop_forever` to `true` (new)
- 2 `Stop Tactic` set `loop_forever` as `true` (changed from 3 to 2)

Enemy Free kick play:
- 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
- 1 `Crease Defender Tactic` is using the default value of `loop_forever`: `true`
- 2 `Shadow Freekicker Tactic` set `loop_forever` as `true`
- 2 `Shadow Enemy Tactic` set `loop_forever` as `true` (Changed from 3 to 2)
- 2 `Move Tactic` set  `loop_forever` as `true`

Kickoff Friendly play:
- 5 `Move Tactic` set  `loop_forever` as `true`
- 1 `Goalie Tactic` is using the default value of `loop_forever`: `true`
- 1 `kickoff Chip Tactic` set  `loop_forever` as `true` (Changed from `Chip Tactic` to `Kickoff Chip Tactic`)

Penalty kick play:
- 1 `Penalty Setup Tactic` set  `loop_forever` as `true` (new)
- 1 `Penalty Kick Tactic` set  `loop_forever` as `true`
- 5 `Move Tactic` set  `loop_forever` as `true`

jonathanlew, 2019-11-09:
Maybe unrelated, but loop_forever be an enum like AutokickType?

garethellis0, 2019-11-10:
Ok @HaomiaoZ - what are your actions moving forwards?

garethellis0, 2019-11-10:
@jonathanlew - I think `loop_forever` makes sense as a boolean because it has a clear positive and negative value. If we were to make it an enum then most booleans in the repo would probably be enums :sweat_smile:

jonathanlew, 2019-11-10:
> @jonathanlew - I think `loop_forever` makes sense as a boolean because it has a clear positive and negative value. If we were to make it an enum then most booleans in the repo would probably be enums

Which would make it much more readable :smile:

MathewMacDougall, 2019-11-10:
Here's my updated list of tactics and plays based on @HaomiaoZ 's latest comment:
**Tactics whose `loop_forever` parameter should be removed**
* GoalieTactic should not expose the `loop_forever` parameter and should default to true
* CherryPickTactic should not expose the `loop_forever` parameter and should default to true
* ShadowEnemyTactic should not expose the `loop_forever` parameter and should default to true
* CreaseDefenderTactic should not expose the `loop_forever` parameter and should default to true
* StopTactic should not expose the `loop_forever` parameter and should default to true
* ShadowFreeKickerTactic should not expose the `loop_forever` parameter and should default to true
* DefenseShadowEnemyTactic should not expose the `loop_forever` parameter and should default to true

**Tactics that should keep the `loop_forever` and have the default value removed**
* MoveTactic should keep the `loop_forever` parameter and the default value should be removed
* PasserTactic should keep the `loop_forever` parameter and the default value should be removed (@garethellis0 would we ever want to loop forever with a passer?)
* ReceiverTactic should keep the `loop_forever` parameter and the default value should be removed (@garethellis0 would we ever want to loop forever with a receiver?)
* ChipTactic should keep the `loop_forever` parameter and the default value should be removed
* PenaltyKickTactic should keep the `loop_forever` parameter and the default value should be removed
* ShootGoalTactic should keep the `loop_forever` parameter and the default value should be removed
* GrabBallTactic should keep the `loop_forever` parameter and the default value should be removed (@garethellis0 do you agree?)
* KickoffChipTactic should keep the `loop_forever` parameter and the default value should be removed
* PenaltySetupTactic should keep the `loop_forever` parameter and the default value should be removed (this tactic will probably get heavily reworked later anyway)

I think the action items now would be to break this up into 1-2 PRs to adjust default values to make everything consistent with what we have above?

garethellis0, 2019-11-10:
> > @jonathanlew - I think `loop_forever` makes sense as a boolean because it has a clear positive and negative value. If we were to make it an enum then most booleans in the repo would probably be enums
> 
> Which would make it much more readable smile

Ahaha, are you saying we should ban the use of booleans?

garethellis0, 2019-11-10:
@MathewMacDougall - sounds reasonable to me, @HaomiaoZ ?

HaomiaoZ, 2019-11-14:
@garethellis0 Agreed with @MathewMacDougall

garethellis0, 2019-11-14:
OK, let's make the change. Should be pretty quick?

HaomiaoZ, 2019-11-15:
should not be too bad as long as all the changes are under `Action` and `Tactic` folder. I am not too sure but can the refactor function in CLion be helpful in this case?

garethellis0, 2019-11-15:
Maybe, not sure. Try it and find out? ¯\_(ツ)_/¯

HaomiaoZ, 2019-11-15:
sure I will give it a try

PR #1064 (2020-01-31): Change `loop_forever` in Action/Tactic

## Please fill out the following before requesting review on this PR

### Description

Change the `loop_forever` parameters to hide the parameter from users or enforce users to specify the value fo `loop_forever` 
### Testing Done

 build and run all bazel software tests after modifications, all tests pass
### Resolved Issues

resolves #767 
### Length Justification

The length is shorter than 500 lines
### Review Checklist
closed 2020-01-31 by HaomiaoZ