UBC-Thunderbots/Software #769: Refactor The Navigator Interface So That It's Generically Testable

garethellis0, 2019-09-20:
### Description of the task
Currently the navigator interface takes intents and return primitives. Unfortunately this means that we can't have a generic set of navigator tests, because we would like to be able check that all navigators do not violate certain constraints in certain scenarios.

We should somehow split up the Navigator interface so that a generic test framework can pass in Intents and get a "Path" back that it can perform checks on.

*This ticket is for planning out the work required for this and creating the needed tickets*

### Acceptance criteria

- [ ] Redesign navigator interface as mentioned above and document here
- [ ] File tickets to implement navigator interface redesign.

### Blocked By

jonl112, 2019-09-29:
@garethellis0 Can we test ThetaStarPathPlanner::findPath instead which takes start and dest and returns a vector of points? Of course it requires that the path planner is initialized with world and obstacles.

MathewMacDougall, 2019-09-29:
We should probably do the same things (test via interface) with the PathPlanners as well, since what @jonl112 said makes sense and it would be good to start with a smaller component. Maybe we can include this in tomorrow's AvoidAreas meeting?

garethellis0, 2019-09-29:
Yes, I generally agree. We should bring it up at the meeting tomorrow.

garethellis0, 2019-10-01:
- Remove the “PlaceHolderNavigator”, rename “PathPlanningNavigator” -> Navigator and move up a level, and just switch out planning algorithms instead
- Use dependency injection to pass a path planner into the “Navigator” constructor
- File structure suggestions:
```
Navigator
Navigator.h
Navigator.cpp
Path_planners
    Path_planner.h
    Straight_line_path_planner.h
    Theta_star_path_planner.h
Obstacle
    Obstacle.cpp
    obstacle.h
```

This could clearly be a few separate tickets, but currently there's no one scoped to work on this and we want to have a bit more design discussion, so holding off on breaking out tickets until then.

**Make sure to to break this down into separate tickets before working on it**

jonathanlew, 2019-10-09:
Proposal: A new path planner interface would define a constructor that will take a `Field` and `findPath` will take `Obstacle`s, start `Point`, and dest `Point` and return a path `std::vector<Point>`. Then ai.cpp will create a path planner and pass it into `Navigator` constructor. `Navigator` can reuse the path planner as much as it likes

garethellis0, 2019-10-09:
*Context for anyone else reading this:* After some offline discussion, we realized that we really only need one navigator, and that what we're really interested in swapping out is the path planning part, as the navigator deals with many other things, like making sure motion rules are obeyed.

garethellis0, 2019-10-09:
@jonathanlew - what if the path planner wanted to return paths in a different, like a parameterized curve? (ex. `path(t) = <x^2 + 1, y>`)? I'm thinking about looking into some spline based navigation this year for smoother/faster motions that account for robot momentum, but these are generally continuous representations (which also means we can more easily pass them over radio).

jonathanlew, 2019-10-09:
@garethellis0 YAGNI for now? If the goal is for ease of testing I'm not sure you could even test the spline and point navigators the same way. Also would a spline navigator would smooth a set of points (e.g. generated by theta star) or generate a curve directly?

garethellis0, 2019-10-09:
Given that this is something I'm planning at looking at fairly soon, and given that continuous representations are an entire class of paths we might want to look at, I think it's important to address this now. Could we maybe return a series of n'th order polynomials (where each polynomial in the series can be a different degree)? That covers quite a lot, and certainly point-to-point navigation.

MathewMacDougall, 2019-10-09:
+1 for a more generic path representation so we can do things like splines or other curves.

I'm not sure however that we would want to force discrete path planners like Theta* or RRT to returned parameterized curves. What we really want is some generic path representation and then the navigator is responsible for breaking that path down into Primitives that get send to the robots.

In the case or Theta*, I imaging the navigator would turn the parameterized curves back into discrete line segments (assuming we don't have spline primitives or anything like that at this point, or don't want to use them in this case). So in this case forcing Theta* to return parameterized curves would be extra work for no benefit.

My current suggestion would be defining a `Path` as an `std::variant` of either line segments or parameterized curves. Then the navigator can handle both types separately and convert them to primitives however it chooses in a generic way.

Thoughts?

garethellis0, 2019-10-09:
That's fine by me, but only because I think a variant with only those two entries covers the _vast_ majority of cases. If we had to add more entries in the variant I'd be more hesitant.

MathewMacDougall, 2019-10-09:
Yes sorry that's more accurate.

jonathanlew, 2019-10-09:
ok sounds good

PR #887 (2019-10-12): New API for findPath for path planner

## Please fill out the following before requesting review on this PR

### Description
New API for findPath for path planner and related navigator, AI, and test changes
Return variant of Curve and points from findPath
Add obstacles and field to arguments for findPath

### Testing Done
Updated regression tests

### Resolved Issues
Resolves #769 

### Length Justification

### Review Checklist
closed 2019-10-12 by jonathanlew