UBC-Thunderbots/Software #770: Create A Generic Test Suite For All Navigators

garethellis0, 2019-09-20:
### Description of the task
Create a generic suite of tests to make sure we can verify that all navigators obey sane behavior. This will likely take the form of designing a series of scenarios that we expect all navigators to be able to handle and writing tests for them. This will allow us to more rapidly bring up new navigation algorithms in the future with much greater confidence.

### Acceptance criteria

- [ ] Implement a series of unit tests that represent the set of scenarios we expect all navigators to be able to handle.
- [ ] Test all navigators with the test suite, fix (or file tickets to fix) any bugs found

### Blocked By

#769

MathewMacDougall, 2020-01-27:
@jordanzliu @garethellis0 @jonathanlew 

Jordan is going to start work on this, but I want to clarify some stuff first since we made this ticket before the navigator re-architecture.

The just want to clarify this will be a generic test suite for `Path Planner` (ie. the part that specifically navigates from point A to B), not any higher-level constructs like `PathManagers`

jonathanlew, 2020-01-27:
> @jordanzliu @garethellis0 @jonathanlew
>
> Jordan is going to start work on this, but I want to clarify some stuff first since we made this ticket before the navigator re-architecture.
>
> The just want [...]

Just testing `Path Planner` sounds good to me. Although we'll eventually want a `PathManager` test, so maybe those can be co-developed or we can make a new issue.

garethellis0, 2020-01-27:
Yeah, I agree. We should work bottom up, and the path planner is the first level from the bottom that does not have a comprehensive set of tests.

PR #1199 (2020-03-30): Path planner test suite

## Please fill out the following before requesting review on this PR

### Description

Create a test suite for path planners

### Testing Done

Tests fail with straight line path planner

### Resolved Issues

Resolves #770 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function def
closed 2020-03-30 by jordanzliu