UBC-Thunderbots/Software #771: Create "Shot" Class To Replace Current Representation

garethellis0, 2019-09-20:
### Description of the task
Currently we use a `std::pair<Point, Angle>` to represent a shot, with the `Point` representing the location to shoot at, and the `Angle` representing the open angle for the shot. We should replace this with a `Shot` class to cleanup the code and allow for easier calculations with them.

A good place to start looking for this is `calc_best_shot.{c,h}`.

Note: @EvanMorcom has some opinions about design for this, so please talk to him as well.

### Acceptance criteria

- [ ] Create a `Shot` class that can be used in place of the `std::pair<Point,Angle>` in all cases where we're considering it a "shot" in our repo
- [ ] Replace instances of `std::pair<Point, Angle>` being used to represent shots in the repo with your new class.

### Blocked By

scveloso, 2019-10-01:
Why is this a class instead of a struct? Is there behavior somewhere else in the code that we want to move to this class?

garethellis0, 2019-10-01:
This could probably be a struct for now, but I seem to recall there being at least a few things that we considered putting in it, and it's much easier to change/update things later on if it's a class rather then a struct.

@EvanMorcom - thoughts?

MathewMacDougall, 2019-10-01:
_Generally_ we have also tried to use classes over structs in our code because we have more control over initialization. For example it's easier to delete the default constructor of a class and force all members to be set with a different constructor, while with structs it can be easier to forget to initialize something.

However we still do use a few structs for really light "wrappers" which is kind of the cases here. Deleting the default constructor is still possible but I did stumble across this: https://stackoverflow.com/questions/33988297/deleted-default-constructor-objects-can-still-be-created-sometimes

I'm not super opinionated either way but also can't remember what else we wanted to put in the "Shot" class...

PR #851 (2019-10-08): Add Shot class and replaced usages of pair<Point, Angle>
### Description
Added ```Shot``` class encapsulating a ```Point``` representing the target of the shot and an ```Angle``` which is the angle representing the obstacles on either side of the shot vector.

### Testing Done
Old unit tests still pass.

### Resolved Issues
resolves #771 

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function definitions (usually in the `.h` file) should have a javadoc style comment at the start of th [... cut]
closed 2019-10-08 by scveloso