UBC-Thunderbots/Software #773: Consider Having Tactics Return Actions

garethellis0, 2019-09-20:
### Description of the task
Currently, tactics return intents by calling actions. However this can make it hard to test tactics, because you have to understand the behavior of the action the tactic is using in order to figure out what intent should be returned. As such, we should consider having tactics return actions, which can then be queried to get the next intent.

This has some issues:
- actions are stateful and use coroutines (so can't be copied easily while running)
- do you _really_ need to understand actions to have an expected output from tactics?

This ticket is to consider this idea, and file tickets if we decide to pursue it.

### Acceptance criteria

- [ ] Decide to pursue this idea or not
- [ ] Figure out work required, if we decide to pursue
- [ ] File appropriate tickets for required work, if we decide to pursue

### Blocked By

MathewMacDougall, 2019-10-18:
Slightly related, while we're redesigning here we should consider how we can make the most guarantees about Tactics and Actions being given data they need. Sometimes right now our `execute` functions rely on data provided by `controlParams`, which means we need to check that data exists before we perofrm operations.

This leads to checks like
```
if(!enemy_threat) {
	// stop
}else {
	// do normal stuff
}
```

Ideally we don't want to perform these checks at all.

garethellis0, 2019-11-05:
I'm going to start poking around at some implementation options for this

garethellis0, 2019-11-09:
Notes from offline slack discussion:

• note down in docs why we don't have setters for `Tactics` or `Actions`
• split all `Action`'s into `setWorldParams` and `setControlParams`
• write a visitor for `Action` 's to update the world params

```tactics = current_play.getTactics()
for (tactic in tactics){
   tacticUpdateVisitor.update(tactic)
}
actions = [tactic.getAction() for tactic in tactics]
for (action in actions){
   actionUpdateVisitor.update(action)
}
intents = [action.getIntent for action in actions]```

<sub>[View in Slack](https://thunderbots.slack.com/archives/D75195G7Q/p1573276588006200)</sub>

jonathanlew, 2020-01-25:
@garethellis0 Is this done?

garethellis0, 2020-01-26:
Almost, some small changes required to wrap it up. Put it on the Feb 8th milestone.

garethellis0, 2020-02-05:
@MathewMacDougall - how strongly do you feel about docs around this? Looking at the software architecture docs this seems like a bit too low level for that doc.

MathewMacDougall, 2020-02-06:
Yea probably too low level for the architecture doc. In-code documentation should be good enough, or just make a ticket to doc it and we can revisit it.

garethellis0, 2020-02-07:
Where in code would we put this? I can't think of a _super_ logical spot.

MathewMacDougall, 2020-02-07:
I think `stp.h` is best, but I've made #1160 to tackle this later in a more comprehensive overview of STP in-code.

PR #1158 (2020-02-07): Moved Updating Of Tactic & Action World Params To Main STP Loop Via Visitors

## Please fill out the following before requesting review on this PR

### Description
Previously, we were setting parameters for actions that were purely derived from the world directly in each tactic. Since we only need the world to update these parameters, this migrates that updating to the main `stp` loop, so that individual tactics only need to worry about updating the bits of actions that are actually specific to that tactic.

Similar deal for tactics, we were previously setting them for each tactic in plays, so [... cut]
closed 2020-02-07 by garethellis0