UBC-Thunderbots/Software #777: Create a templated Visitor design pattern

MathewMacDougall, 2019-09-20:
### Description of the task

We are starting to use the Visitor design pattern more, for example with radio / primitives, and soon to be tactics and parameters. As such, we should make a generic templated visitor so we don't have to repeat so much work to set up the design pattern.

Create a new `visitor.h` template under a new `util/design_patterns` folder and implement the templated visitor from https://stackoverflow.com/questions/55801803/how-to-write-includes-with-visitor-pattern-simple-example

### Acceptance criteria

- [ ] Implement the templated visitor
- [ ] Test the visitor implementation

### Blocked By

garethellis0, 2019-11-12:
After trying this, I really don't think it's much of an improvement. Long story short, if we went with what everyone seems to suggest in terms of templating (at least what I could find), we would define a visitor like:
```
class PrimitiveVisitor : public Visitor<CatchPrimitive, ChipPrimitive, DirectVelocityPrimitive, etc.> {}
```
which is definitely more succinct, but we wouldn't really be able to make the `Primitive` implementation side of things simpler without significant extra complexity.

Long story short, I don't think that using templating here gives a significant enough benefit to warrant this change.

I would seriously encourage reading the first answer on https://stackoverflow.com/questions/11796121/implementing-the-visitor-pattern-using-c-templates, it kind of shows how it would work, and hopefully that will help illustrate my point.

@jonathanlew @akhilveeraghanta @MathewMacDougall

jonathanlew, 2019-11-12:
That makes sense. On another note, a annoying overhead is updating the all_tactics, all_intents, etc files and the associated build objects. Can we automate that?

garethellis0, 2019-11-12:
Definitely, I think #865 covers that? Lemme know if I'm not quite understanding what you mean here though!

jonathanlew, 2019-11-12:
ah yes, I forgot I filed an issue for that

MathewMacDougall, 2019-11-13:
Oh neat I didn't see that ticket either. After reading the post I generally agree this doesn't seem worth it.

closed 2019-11-13 by garethellis0