UBC-Thunderbots/Software #779: Fix All Compiler Warnings

garethellis0, 2019-09-20:
### Description of the task
With our migration to bazel, we have started to get a _lot_ of compiler warnings that we just didn't have turned on before. We should fix them all and make warnings fail compilation.

### Acceptance criteria

- [ ] Make warnings errors (so they will fail compilation)
- [ ] Fix all warnings in the codebase

### Blocked By

garethellis0, 2019-10-17:
While updating some toolchain stuff for https://github.com/UBC-Thunderbots/Software/issues/849, namely updating our compiler to clang instead of gcc. I'm getting some warnings that should definitely have caused errors with `-Werror`. I _think_ what might have happened is that some cached build outputs were not properly rebuilt when the flags were changed. @scveloso, can you:
- disable the remote cache by commenting out the line `build --remote_http_cache=https://storage.googleapis.com/thunderbots_bazel_cache_us_multi_region` in `src/.bazelrc`
- run `bazel clean --expunge` in `src`
and then see if there are any compiler errors that we should fix? If so we should re-open this.

garethellis0, 2019-10-17:
One example is that `point` is a unused variable on line 171 of `polygon_test.cpp`:
```
TEST(PolygonTest, test_polygon_triangle_intersects_ray2)
{
    Point p1{0.0f, 0.0f}, p2{1.0f, 0.0f}, p3{1.0f, 0.0f};
    Point point{2.0f, 2.0f};
    Polygon triangle{p1, p2, p3};

    Ray ray{Point(-0.1f, -0.1f), Vector(-1.0f, -1.0f)};
    EXPECT_FALSE(triangle.intersects(ray));
}
```

garethellis0, 2019-10-17:
Ok, in order to limit scope on #849 (it's already really big), I'm going to re-open this ticket. The current scope is as follows:
- disable all caching and clear the bazel cache, as detailed above
- enable warnings as errors by adding `"-Werror"` to the `warnings_feature` in `cc_toolchain_config.bzl`
- enable the `-Winvalid-constexpr` warning by removing `-Wno-invalid-constexpr` from the `warnings_feature` in `cc_toolchain_config.bzl`
- figure out how to disable warnings for external packages, like `gtest`, `boost`, `g3log`, etc.

@scveloso - if this seems reasonable to you, could you please update the description for this ticket accordingly?

garethellis0, 2019-10-17:
It's also possible we've gained a few more error flags in the `gcc`->`clang` migration. If that's the case then we can probably file a new ticket, since this one was just for the `gcc` errors.

scveloso, 2019-10-17:
@garethellis0 this is reasonable. I'll take a closer look later and update the issue as soon as possible.

garethellis0, 2019-11-02:
Ok, looks like the new toolchain catches some issues the old one doesn't. My comment above stands, re-opening.

@scveloso - no need for you to work on this right now, would rather you focus on geom, this can go back into the ticket pool.

garethellis0, 2019-11-03:
@amrm3lm - just as an additional note, just do a quick search over the codebase for this issue number (779), I think we've tagged a few places stuff needs to be changed once this is fixed.

PR #858 (2019-10-10): Fix compiler warnings and treat warnings as errors
### Description
- Fix all compiler warnings
- "Treat warnings as errors" setting added in bazelrc

### Testing Done
- Bazel build should fail when encountering any warnings
- Old unit tests still pass

### Resolved Issues
resolves #779 

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function definitions (usually in the `.h` file) should have a javadoc style comment at the start of them. For examples, see the functions defined in `thund [... cut]
PR #1051 (2020-01-21): Fix All Compiler Warnings

## Please fill out the following before requesting review on this PR

### Description
added warning as error compiler option and fixed all warnings that resulted. 

Because of the way we designed our primitives and intents, we had lots of "-Woverloaded-virtual" warnings for overloading virtual functions. Because there are many instances of this warning and no way to easily modify the code to fix it (to my knowledge), i ended up suppressing this specific warning. 

Some performance tests contained unused variables (ex cost_function_test.cpp ) , but thats only because  [... cut]
closed 2020-01-21 by amrm3lm