UBC-Thunderbots/Software #793: Re-think how AvoidAreas should work

MathewMacDougall, 2019-09-20:
### Description of the task

Now that we better understand our use cases for AvoidAreas, how we use them, and how we (think) we want to control them we should rethink the idea in general. They work well and the idea of discrete areas to avoid on the field makes sense, but how they were set and communicated between the strategy and the navigator got too complicated.

We should figure out how we want to solve this. New datastructures? New concept entirely?

One suggestion was to replace the AvoidArea enums with classes and inheritance (a class for each discrete type of area).
* Abstract `AvoidArea` class, and concrete implementation of each "type", like area around ball.
* The classes could then provide `getAvoidArea` or `getShape` function that can be consumed by the navigator generically.
* The classes could also be constructed with parameters if Tactics or Actions need more direct control

### Acceptance criteria

- [ ] Make a plan for how to handle AvoidAreas

### Blocked By

garethellis0, 2019-10-01:
Notes from 2019-09-30 meeting:
- Rename “AvoidAreas” to instead represent rules & constraints on rules
- Remove all “AvoidAreas” that are NOT related to rules
- Avoiding robots based on the rules is the responsibility of the navigator, the actions/intnets should have no knowledge of it
- MoveIntent will have a flag for touching / not touching the ball, because there is semantic information in the high level about whether to touch the ball or not
- No more whitelist/blacklist in the tactic or play. Instead, have a visitor that produces a list of AvoidAreas/RuleConstraints as a function of a tactic and a gamestate ((Tactic,GameState)->std::vector<RuleConstraints>).
- In all cases where there is a play whitelisting something on a tactic, we probably just need another tactic (for example, kickoff)
- STP manager will getruleconstrainsts and then apply to intent from tactic
- Ball flag for moveintent

jonathanlew, 2019-10-08:
@MathewMacDougall @garethellis0 I scoped out the work in #874 and #875. Can you review that and let me know if I close this issue?

MathewMacDougall, 2019-10-09:
Left a comment on #874 but otherwise they seem fine. Maybe we should still leave this ticket open until the others are completed in case more work is broken out?

garethellis0, 2019-10-09:
I think #875 supersedes this so we can close this one.

closed 2019-10-09 by jonathanlew