UBC-Thunderbots/Software #796: Move Action looping logic to superclass

MathewMacDougall, 2019-09-20:
### Description of the task

Right now Actions implement their `loop_forever` behavior by using a `while(loop_forever && some_other_conditions)` loop in their implementation. Ideally developers shouldn't have to worry about the `loop_forever` part of this logic and re-implement it in every Action. Instead, it should be handled in the superclass like how Tactics do it (managing the coroutines etc.)

### Acceptance criteria

- [ ] Handle Action looping logic based on `loop_forever` in the Action superclass
- [ ] Test that this looping logic works properly.

### Blocked By

amrm3lm, 2020-04-11:
@MathewMacDougall would this apply to actions that have a `while(true)` in their implementation, effectively making them loop forever? Ex: chip_action, kick_action, pivot_action

The way i see it is that `loop_forever` is meant to indicate the action being completed and restarted, whereas a `while(true)` means the action needs to constantly reevaluate itself and to ensure that we keep the looping logic in the subclass.

MathewMacDougall, 2020-04-11:
Yes my initial idea was that it would apply to those Actions too. The `while(true)` effect can be effectively reproduced by hardcoding `loop_forever=true` when the specific Action calls the Action superclass constructor, similar to how some Tactics do it.

Any actions that are currently implemented with `while(true)` shouldn't expose the `loop_forever` parameter at all.

PR #1261 (2020-04-15): Move Action looping logic to superclass

## Please fill out the following before requesting review on this PR

### Description

- moved loop_forever logic to action superclass
- this includes actions who are always implicitly looping (eg chip_action, kick_action, pivot_action)

### Testing Done

Passes existing tests. 
Also added loop_forever parameter to `move_test_action` and tested with it.  
### Resolved Issues

resolves #796 
### Length Justification

### Review Checklist
closed 2020-04-15 by amrm3lm