UBC-Thunderbots/Software #799: Move thunderbots/* UP One Level

garethellis0, 2019-09-20:
### Description of the task
Currently we have a folder structure that looks like `src/thunderbots/...`. There are other folders under `src`, but these should all be removed once ROS is gone. We should move everything in `src/thunderbots/` to `src/`.

### Acceptance criteria

- [ ] Remove any remaining files in `src/` except for `src/thunderbots`
- [ ] Move everything from `src/thunderbots/` to `src/` and delete `src/thunderbots`
- [ ] Fix any scripts that are dependent on these paths (check `travis_ci.sh`, everything in `clang_format`, everything in `environment_setup`).

### Blocked By

jonathanlew, 2019-10-08:
@garethellis0 is this done?

MathewMacDougall, 2019-10-08:
yes this was done as part of the bazel migration PR

closed 2019-10-08 by MathewMacDougall