UBC-Thunderbots/Software #802: Make a new mrftest

MathewMacDougall, 2019-09-20:
### Description of the task

In order to support MECH and ELEC development, as well as our software / firmware development, we need a new tool that will allow us to control the robots at a low level without needing the AI or vision.

In the past we had an application called `mrftest` that was a simple GUI showing robots state, and allowed for basic control like charging the capacitors, moving individual wheels, or telling the robot to drive in a direction. This was very useful for testing and debugging robots electromechanically.

It will be critical we re-implement a tool like this early this year in order to support development. A more concrete plan and actions items should be made.

### Acceptance criteria

- [ ] Collect desired features from MECH and ELEC
- [ ] Re-implement `mrftest` in some capacity

### Blocked By

jonathanlew, 2019-12-01:
Needs to send primitives, too

akhilveeraghanta, 2020-02-02:
@jonathanlew I think I might have some extra bandwidth this week I can take this on if ur swamped with tickets, it also helps me test that the multicast groups are working and really make everything with proto support is ironed out

jonathanlew, 2020-02-03:
> @jonathanlew I think I might have some extra bandwidth this week I can take this on if ur swamped with tickets, it also helps me test that the multicast groups are working and really make everything with proto support is ironed out

Go for it if you can!

MathewMacDougall, 2020-03-10:
Reassigning to Jon since he's currently working on this

PR #1227 (2020-04-06): Robot Diagnostics GUI

## Please fill out the following before requesting review on this PR

### Description
Created robot_diagnostics_main executable with a completed GUI and stubs for integrating with robot_communicator or other parts of software. This is a replacement for mrftest and is a standalone executable that will be integrated with software in the future.

Run with `bazel run //software/robot_diagnostics:robot_diagnostics_main`

[image]

### Testing Done
Manual testing of GUI

### Resolved Issues
Resolves #802 

### Length Justification
Don't see any benefit to breaking up the bits o [... cut]
closed 2020-04-06 by jonathanlew