UBC-Thunderbots/Software #811: Design a New Dynamic Parameter System

garethellis0, 2019-09-21:
### Description of the task
Our current parameter system is highly monolithic and not very modular, which makes it difficult to test. We should design a new one. 

NOTE: most of the design work for this is being tracked in [this](https://docs.google.com/document/d/1MX824V1DC7oCQixdbVqfsLsnR9JzGFPCrLycQDi0uNE/edit?usp=sharing) doc.

### Acceptance criteria

- [ ] File all required tickets in order to implement the new Dynamic Parameter system

### Blocked By

garethellis0, 2019-12-24:
@akhilveeraghanta - do we have tickets filed for the rest of the work left here? If not could you please create some? Either way I think we can close this now?

garethellis0, 2020-01-06:
@akhilveeraghanta

MathewMacDougall, 2020-04-21:
Closing as V2 is complete. For future work see #1295 #1296 #1297 #1298

closed 2020-04-21 by MathewMacDougall