UBC-Thunderbots/Software #812: Clean Up Obstacle & Robot Obstacle Classes

garethellis0, 2019-09-21:
### Description of the task
Cleanup and improve the `Obstacle` and `RobotObstacle` tests. This is a collection of suggested improvements, see "Acceptance Criteria" below. 

### Acceptance criteria

- [ ] use a factory to create obstacles from Robot, Ball, etc. There should be no need to compile those things in to the obstacle class
- [ ] `RobotObstacle` should extend `Obstacle`
- [ ] `Obstacle` should just extend some sort of shape type (this depends on #781) instead of implementing it's own geometry utils
- [ ] We should remove all "magic" scaling factors from both `Obstacle` and `RobotObstacle` and instead choose values grounded in real-world values.

### Blocked By

#781

jonathanlew, 2019-10-27:
We should be using sets of obstacles instead of vectors, but we don't support >, ==, <, etc operators

garethellis0, 2019-10-27:
Sorry, why do we need `<` and `>` for a set? And can we just add `==` operators?

jonathanlew, 2019-10-27:
Sorry I'm not being precise here. Whatever it is we need to make it a set, we should implement that

MathewMacDougall, 2019-10-27:
Oh. Usually just a hash function for `unordered_set`, and I think just `<` for `set` (which is implicitly ordered). But yea sounds good.

jonathanlew, 2019-10-27:
unordered_set sounds good, we don't care about any ordering at all (which is why vector makes no sense)

garethellis0, 2019-10-28:
Ah I see. Thanks for noting that, this discussion will be helpful for whomever takes this issue.

jonathanlew, 2019-11-04:
>     * use a factory to create obstacles from Robot, Ball, etc. There should be no need to compile those things in to the obstacle class
@garethellis0 Can this factory also create obstacles from motion constraints and cover the work for #1010?

garethellis0, 2019-11-05:
That seems reasonable to me at first glance.

PR #1320 (2020-05-19): Integrated New Obstacles and Refactor ObstacleFactory

## Please fill out the following before requesting review on this PR

### Description
Integrated obstacles from new_obstacles :
* Cleaned up obstacle factory 
* Replace Obstacle with ObstaclePtr
* Remove VelocityObstaclePathManagerConfig

### Testing Done
New unit tests and updated tests

### Resolved Issues
Resolves #812 

### Length Justification
Sorry, there's a lot of boiler plate and new integration involved with the new hierarchy and a lot of terrible code that had to be deleted

### Review Checklist
closed 2020-05-19 by jonathanlew