UBC-Thunderbots/Software #813: Ensure All Geom Functions Appropriate Use `const&`

garethellis0, 2019-09-21:
### Description of the task
Over the past year we've discovered that several geom functions were taking things by value instead of by const reference. This can result in a significant slowdown, particularly as these functions are called quite often. We should remove all the functions in `geom` to see if they're using `const&`, and if not, change them if they should be.

### Acceptance criteria

- [ ] Review all functions in `software/geom` and change them so they take values by `const&` wherever **safely** possible

### Blocked By

scveloso, 2019-10-05:
> We should remove all the functions in geom to see if they're using const&, and if not, change them if they should be.

Sorry, what does this mean? How will removing functions in geom tell us if they're using const&? Or is this a typo?

garethellis0, 2019-10-05:
Sorry yup, that's a typo. Should be `We should check all the functions ...`

PR #1618 (2020-07-18): Geometry Clean Up and Folder Restructure

## Please fill out the following before requesting review on this PR

### Description
Cleaned up the last bit of geom/util, miscellaneous geom, and moved folders around

### Testing Done

### Resolved Issues
resolves #959
resolves #813 
resolves #1593 
resolves #1494 
resolves #1444

### Length Justification
Hard to break this up

### Review Checklist

**_It is the reviewers responsibility to also make sur
closed 2020-07-18 by jonathanlew