UBC-Thunderbots/Software #829: Setup CI To Automatically Upload Build Results To Our Bazel Cache

garethellis0, 2019-09-27:
### Description of the task
When our code is built, we have the ability to upload the compiled artifacts to a remote server that anyone can read from, in order to help speed up other peoples builds and/or CI. We would like to upload the build artifacts from the CI build to this server, since that would mean that if the code is on `master`, anyone pulling new code wouldn't have to recompile it (they could just ask the server for it. See the [bazel remote caching doc page](https://docs.bazel.build/versions/master/remote-caching.html) for more details.

In order to upload to our bazel cache a secret keyfile is required, because we want to restrict who can write to it. Since we can't commit the key to our public repo (because then _anyone_ would be able to view it), we need some way to provide it to CI. One option could be to use [travis CI encrypted keys](https://docs.travis-ci.com/user/encryption-keys/). Note that these environment variables are not available for PR builds (since then anyone could submit a PR to expose them). However I _think_ that there is a clean build done on `master` after each PR merges, in which these keys _are_ available.

Ask @garethellis0 for more details and how to get access to the remote cache.

### Acceptance criteria

- [ ] All CI builds running on `master` (not neccesarilly all CI builds for PR's) should upload all compiled artifacts to our remote cache,

### Blocked By

MathewMacDougall, 2019-10-20:
Reverted this change so re-opened.

garethellis0, 2019-10-20:
I'm dropping this for now because I have limited scope, anyone can feel free to take it.

For context, we merged this in https://github.com/UBC-Thunderbots/Software/commit/1ba50861ad9d913c2ac6bd027ae127e0b1261977, but had CI failures on master, so had to revert it in https://github.com/UBC-Thunderbots/Software/commit/15168bf5d5ed17fb6865708cc5bcb53dc256a5ee

garethellis0, 2020-01-06:
After much work, we've concluded that a remote cache is not worth the effort.

PR #953 (2019-10-19): Uploading To Bazel Remote Cache From CI

## Please fill out the following before requesting review on this PR

### Description
This should allow CI to upload artifacts to the remote cache when it runs on `master`.

### Testing Done
Ci passes, but really this needs to run on `master` for the real test :grimacing: 

### Resolved Issues
resolves #829

### Length Justification
N/A

### Review Checklist
closed 2020-01-06 by garethellis0