UBC-Thunderbots/Software #832: Control Robot with XBox

MathewMacDougall, 2019-09-28:
### Description of the task

We would like to be able to control a single robot using an XBox controller. This is very useful for demos, and sometimes testing (it's also fun).

We would like to implement XBox control with our new software system. This will require:
1. Finding a library to read XBox controll input over USB
2. Creating an executable that will transform the XBox controller data into Primitives that will be sent to the robot. The backend can handle the sending the primitives, the xbox module just needs to handle translating input into primitives
3. We have a DirectVelocity primitive that can be used for this, as it's easy to map a joystick direction to a velocity vector.

### Acceptance criteria

- [ ] Find a library that can read XBox controller input
- [ ] Create an executable that converts the XBox input into primitives and sends those commands to a specified robot

### Blocked By

MathewMacDougall, 2019-11-03:
A dribbler motor blew at a demo recently (still unsure if it got too hot or was another failure). Henry had a good idea as a result.

Currently we toggle the dribbler by pressing the "Y" button. We would like to change this to only keeping the dribbler on if a button is held down (maybe a trigger instead of "y"), so that the dribbler won't stay on for as long or be forgotten and left on. This should hopefully minimize any issues like this in the future when doing demos.

MathewMacDougall, 2019-11-11:
Hey @Solaris5959 just checking in on where you're at with this. The MECH team has been using the xbox a lot more to test new chassis and robot stuff, so this is becoming higher priority since we can't just keep using the old version on my laptop.

Solaris5959, 2019-11-16:
Removed myself as I am unable to commit enough time to complete the project, sorry. XInput library seems to be the most popular used for XBox controller input. Shouldn't be too difficult to map primitives to the inputs once you have the class for input set up.

MathewMacDougall, 2019-11-17:
Alright, no worries. Thanks for letting us know about the library.

jonathanlew, 2020-04-17:
@MathewMacDougall Has this been resolved by #1121?

MathewMacDougall, 2020-04-18:
Yea I'd call this complete

closed 2020-04-18 by MathewMacDougall