UBC-Thunderbots/Software #834: Investiage init order issue in full_system_main

MathewMacDougall, 2019-09-28:
### Description of the task

In 	full_system_main.cpp`, the order in which the `ai` and backend are initialized affect system behaviour. If the `ai` is initialized before we parse the command line args (where the backend is initialized), the system works as expected. However if the `ai` is initialized after the command line args / backend, the system will crash when the AI gets an invalid refbox gamestate.

The refbox gamestate looks like an uninitialized value, so first check it's being initialized properly. Try figure out why this bad value is only gotten in one case vs the other. Ideally the order shouldn't matter

### Acceptance criteria

- [ ] Fix the bug

### Blocked By

jonathanlew, 2020-04-07:
@MathewMacDougall any update?

MathewMacDougall, 2020-04-07:
Not sure. I seem to recall someone looking into this, not sure if it's still an issue or not. I can take a look later, but won't get to it for at least a few days due to capstone.

PR #1526 (2020-06-27): Remove unneeded namespaces

## Please fill out the following before requesting review on this PR

### Description

We still had some lingering namespaces we could get rid of, so I did. I ignored anything in the old radio or USB code. Removed:
* The Util namespace
* The Constants namespace
* The GeomConstants namespace
* a few anonymous namespaces

### Testing Done

unit tests still pass

### Resolved Issues

this happens to resolve #834 (or at least I confirmed it's no longer an issue).

### Length Justification

### Review Checklist
closed 2020-06-27 by MathewMacDougall