UBC-Thunderbots/Software #839: Setup Clang-Tidy In Our Repo

garethellis0, 2019-09-28:
### Description of the task
`clang-tidy`(https://clang.llvm.org/extra/clang-tidy) is a super useful tool with a variety of linting tools with automatic fixes for some things. It would be super helpful if we could have this perform automatic checks on the codebase (as part of CI) to save reviewers time commenting on little nitty-gritty things (and hopefully make it easier for PR submitters to fix them).

### Acceptance criteria

- [ ] Setup clang-tidy with options that best match our coding conventions
- [ ] Integrate `clang-tidy` into `fix_formatting.sh` or write a separate script for it
- [ ] Made `clang-tidy` a blocking check in CI
- [ ] Fix any `clang-tidy` issues found

### Blocked By

garethellis0, 2019-09-28:
This may by useful: https://github.com/aszinovyev/bazel-compilation-db

jonathanlew, 2019-10-11:
Can this run when we merge our code into master so PRs don't get failed builds?

garethellis0, 2019-10-11:
Modifiying code in CI is generally a poor/dangerous idea. Generally the way I've seen this done is that a clang-tidy check is made part of the pre-commit git hooks so that all commits are checked before they're created on the users machine.

Also, I'm not sure that all clang-tidy errors (like variable naming) can be automatically fixed.

jonathanlew, 2019-10-11:
pre-commit git hooks super annoying if they take a while to run...

garethellis0, 2019-10-11:
Yeah, I agree it's not ideal, but making modifications to code that aren't reviewed at all is just a really bad idea. Worst case it's one more commit at the end of a PR.

garethellis0, 2019-11-21:
This is extremely low priority right now and possibly of minimal benefit. Likely not getting done this year so closing.

closed 2019-11-21 by garethellis0