UBC-Thunderbots/Software #874: Create new tactics so each tactic corresponds one whitelist

jonathanlew, 2019-10-08:
### Description of the task
The new `MotionConstraint` system requires that each `Tactic` only corresponds to one whitelist. Some `Tactic`s have stages with different whitelists, and they may need to be broken up and the plays adjusted to respect this requirement.
See more details in #793 and #854

### Acceptance criteria

- [ ] Identify all `Tactic`s that use multiple whitelist and break them up
- [ ] Adjust the plays to accommodate
- [ ] Testing to ensure nothing is broken

### Blocked By
Migrating tactics from Robocup2019

MathewMacDougall, 2019-10-09:
I don't totally understand what you mean by each tactic corresponding to one whitelist? Do you just mean each tactic is associated with a different list, since the tactics are independent?

MathewMacDougall, 2019-10-09:
Also does this ticket include implementing the visitors?

jonathanlew, 2019-10-09:
> Also does this ticket include implementing the visitors?

Yes, for the new tactics

jonathanlew, 2019-10-09:
> I don't totally understand what you mean by each tactic corresponding to one whitelist? Do you just mean each tactic is associated with a different list, since the tactics are independent?

Each tactic, no matter its constructor or state or any other internal variable, corresponds to a single whitelist

We might expect that most tactics should share the same whitelist, which is the empty set.

garethellis0, 2019-10-09:
@jonathanlew - just to make sure I understand this fully:
we've defined a mapping `(GameState, Tactic) -> MotionConstraints`, but there are some cases (like during kickoff) where we have a different set of tactics depending on the play stage we're in?

If that's the case, I'm struggling to think of any examples. In any case I can think of where we would change the `MotionConstraints` on a given tactic, it's because of a `GameState` change. Can you or @MathewMacDougall find any counter-examples?

MathewMacDougall, 2019-10-09:
@garethellis0 you're specifically looking for (counter)examples where a Tactic would want to change it's whitelist in the middle of execution for a reason unrelated to a `gamestate`?

I have no current counterexamples and we didn't come up with any during our initial meeting. IIRC if we ever came up with a specific case like that, the thing being avoided would be turned into an `Intent` parameter (like we're planning on doing with the ball), or possibly something else if we're way outside the box.

jonathanlew, 2019-10-09:
@MathewMacDougall @garethellis0 I'd have to look at the migrated tactics to know for sure. The other thing is when plays pass different whitelists into Tactics depending on the Play's state, that that also breask the ` (GameState, Tactic) -> MotionConstraints` mapping. This issue needs to fix those cases too.

MathewMacDougall, 2019-10-09:
And just to clarify, our solution was to stop Plays passing whitelists into Tactics right?

garethellis0, 2019-10-10:
Correct, there will only be a `(GameState, Tactic) -> MotionConstraints` mapping now.

PR #994 (2019-10-30): Consolidate avoid areas from plays into tactics except for BALL

## Please fill out the following before requesting review on this PR

### Description
We want to move avoid area differentiation to the tactic level in preparation for motion constraints, so we're consolidating avoid areas from plays into tactics except for BALL. Each tactic now corresponds one whitelist.

### Testing Done
Regression tests

### Resolved Issues
Resolves #874

### Length Justification

###
closed 2019-10-30 by jonathanlew