UBC-Thunderbots/Software #951: Bug: When Running With GrSim Backend From Command Line, Receive Garbage World

jonathanlew, 2019-10-17:
### Description of the task
#### Bug Description:
Log:
```
Setting current_game_state to  1658124368
Calling name on 1658124368
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Error: Tried to get the name of an invalid refbox gamestate
```
getIntents sets current_game_state to 1658124368, then getPlayInfo calls name on 1658124368 and gets thrown std::invalid_argument
This means that onValueReceived got a garbage world and it called getIntents with garbage data. Garbage in garbage out
#### How to reproduce:
Run `bazel run //software:full_system -- --backend=grsim` from command line, then run `grsim` from a different terminal. It should crash as soon as grsim loads.

### Acceptance criteria

- [ ] Fix the bug and manually test

### Blocked By

MathewMacDougall, 2019-10-17:
Confirmed failing behavior on my machine.

garethellis0, 2019-10-18:
@jonathanlew - could you confirm if this is a bug on commit `ac882d3df0aa0d108e515`? That's an important one because it's when we merged the new visualizer and bazel stuff. I tested several times during that process, but it's entirely possible I missed some last minute changes.

jonathanlew, 2019-10-19:
@MathewMacDougall @garethellis0 This is _not_ a bug for ac882d3df0aa0d108e515 :thinking:

MathewMacDougall, 2019-10-20:
Oh.....

garethellis0, 2019-10-20:
That's good. Can you use `git bisect` to narrow down what commit caused the regression? @MathewMacDougall - perhaps we should consider a full system test that runs the visualizer, even if it can't do anything, and makes sure that nothing segfaults at a minimum.

MathewMacDougall, 2019-10-20:
That's only meaningful if we can also run grsim or something (maybe simulator backend soon), at least if you want to catch issues like this.

jonathanlew, 2019-10-20:
I will git bisect

jonathanlew, 2019-10-20:
6486de97e601e50c0e2eb92ae005aa63748afcf3 is the culprit

jonathanlew, 2019-10-20:
@henryymliu Any ideas?

jonathanlew, 2019-10-20:
we're not destroying the pass generator anymore...

garethellis0, 2019-10-20:
I assume you're referring to this line: https://github.com/UBC-Thunderbots/Software/commit/6486de97e601e50c0e2eb92ae005aa63748afcf3#diff-b5df6e3839f3a46d339d32d06a8bb6ddL184? This change is correct, manually destructing the pass generator like this is a bug, because since it is just a normal variable, the compiler will attempt to destruct it when it leaves scope, and calling the destructor twice on an object results in undefined behavior. This is what cause the robots to just stop moving during corner kicks in our first few games at competition.

garethellis0, 2019-10-20:
> That's only meaningful if we can also run grsim or something (maybe simulator backend soon), at least if you want to catch issues like this.

Good point, definitely something to consider for after the simulator is finished then.

garethellis0, 2019-10-20:
I am quite confused about this one. It seems like all the changes in 648de9 are downstream of where the game state would be set. This _could_ be a more subtle bug with how the actual binary is compiled. 

If we can't get this figured out ASAP, how feasible is it to revert this commit? @MathewMacDougall - I think you'd be the only one with dependencies on these changes, though the other conflicts probably won't be pretty. @henryymliu - FYI since that was your commit (also not pointing fingers here, this is a _super_ subtle bug, not catching things like this is a failure of our process, not a fault of anyone in particular).

garethellis0, 2019-10-20:
OK I think I've narrowed it down. It seems like however we're constructing the `World` in `network_client.cpp`, the game state is being initialized to a garbage value. This bug is dependent on `filterAndPublishVisionData` being called before `filterAndPublishGameControllerData`, because both will publish the world, but `filterAndPublishGameControllerData` will update the game state to a none-garbage value before doing so.

Basically we just need to figure out why we're initializing the game state to a garbage value, and then fix that. Looking into it now.

garethellis0, 2019-10-20:
And this is why we write "simple" tests for stuff like default constructors :stuck_out_tongue: . This test fails:
```
TEST(GameStateTest, default_constructor){
    GameState game_state;

    EXPECT_EQ(RefboxGameState::HALT, game_state.getRefboxGameState());
}
```
with
```
TEST(GameStateTest, default_constructor){
    GameState game_state;

    EXPECT_EQ(RefboxGameState::HALT, game_state.getRefboxGameState());
```

MathewMacDougall, 2019-10-20:
Yee haw. I guess this falls into my domain, so I'll take this since Jordan is already busy.

MathewMacDougall, 2019-10-20:
Looks like you've made good progress on this. Would prefer to fix now rather than revert.

garethellis0, 2019-10-20:
Yeah absolutely agreed. Reverting would likely be very painful.

PR #967 (2019-10-20): Properly Default Initialized game_state In GameState

## Please fill out the following before requesting review on this PR

### Description
- properly initialized `game_state` in `GameState` so that we don't get garbage values if we try to read it before calling `updateRefboxGameState`
- added a test for default construction of the `GameState` class
- tweaked some naming to be a bit more consistent

### Testing Done
Added unit test, confirmed #951 is resolved.

### Resolved Issues
resolves #951

### Length Justification
closed 2019-10-20 by garethellis0