UBC-Thunderbots/Software #957: Implement Splines, Rays and Polynomial hierarchy

scveloso, 2019-10-19:
### Description of the task
Implement the following hierarchy as part of the geom redesign:
- Spline <- Line segment
- Ray
- Polynomial <- Line

These classes/interfaces should only have fields, getters and setters.
Methods that operate on these belong in util and they will be in a separate issue.

### Acceptance criteria

- [ ] Hierarchy implemented as above
- [ ] Each class has the fields, getters and setters it used to have before

### Blocked By
- #955
- #956

garethellis0, 2019-11-12:
Just increasing the difficulty on this one a bit. We should probably doc the criteria for each difficulty level. Opened https://github.com/UBC-Thunderbots/Software/issues/1032 for myself to do that.

jsn0716, 2020-02-01:
Only things left are contained in smaller issues #983 and #1098, closing this issue now

closed 2020-02-01 by jsn0716