UBC-Thunderbots/Software #959: Implement geom util functions

scveloso, 2019-10-19:
### Description of the task
As a part of the geom redesign, implement the currently used geom functions in a util file and try to use boost geometry as much as possible, especially for difficult computations.

### Acceptance criteria

- [ ] All old methods that are being used are reimplemented/considered

### Blocked By
- #958 
- #957 
- #956

jonathanlew, 2019-11-04:
@jsn0716 @scveloso Can we include #209, #781, and #398 in the scope of this issue and close the other issues? Can we update the description to include these extra details? @garethellis0 @MathewMacDougall thoughts?

garethellis0, 2019-11-04:
@jonathanlew - agreed, happy to consolidate all mentioned issues into this one

MathewMacDougall, 2019-11-04:
Fine with me as long as we don't end up with 1 giant super-PR.

garethellis0, 2019-11-05:
Yeah, @jsn0716 @scveloso - when you get to this please break it up. There a _many_ single geometry functions that are probably worth an issue on their own!

jsn0716, 2019-11-05:
Sounds good to me, I'll keep them in this issue for now and once @scveloso and I get to it, we can break it  up into smaller issues as needed

PR #1618 (2020-07-18): Geometry Clean Up and Folder Restructure

## Please fill out the following before requesting review on this PR

### Description
Cleaned up the last bit of geom/util, miscellaneous geom, and moved folders around

### Testing Done

### Resolved Issues
resolves #959
resolves #813 
resolves #1593 
resolves #1494 
resolves #1444

### Length Justification
Hard to break this up

### Review Checklist

**_It is the reviewers responsibility to also make sur
closed 2020-07-18 by jonathanlew