UBC-Thunderbots/Software #982: Make GrSim-Visualizer Run Offline

jonathanlew, 2019-10-22:
### Description of the task
Currently, GrSim uses UDP to communicate with visualizer and it uses some sort of wifi/ethernet port of our computers to do that. This means that if we aren't connected to wifi/ethernet, then GrSim cannot communicate with visualizer and it also crashes because it can't find a port, I guess.

This was an issue in Robocup 2019 because we wanted to test plays/tactics on grsim/visualizer on the plane ride over to Sydney.

### Acceptance criteria

- [ ] Make GrSim Visualizer Run Offline

### Blocked By

MathewMacDougall, 2020-03-10:
Don't think this is a high priority, but should be easy to at least determine if it's possible or not.

@jonathanlew when you mean offline, you mean just not connected to a network? Or your local interfaces disabled / completely turned off?

@akhilveeraghanta is it possible for local multicast to work while not connected to an external network? Would you just need to specify loopback?

jonathanlew, 2020-05-27:
@MathewMacDougall Will the new simulator be able to work offline? (i.e. not connected to a wifi/ethernet, but local interfaces on)

MathewMacDougall, 2020-05-27:
AFAIK yes, but lets leave this open as a reminder until then.

PR #1572 (2020-07-05): Remove support for GrSim

## Please fill out the following before requesting review on this PR

### Description

Remove grsim and a few related hacks we had.

### Testing Done

Code builds, can still use other backends as usual.

### Resolved Issues

resolves #1416
resolves #982
resolves #742
resolves #1281 

### Length Justification

Yeet

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been cov
closed 2020-07-05 by MathewMacDougall