UBC-Thunderbots/Software #983: GEOM: Make a Spline(std::vector<Points> points, unsigned int order) constructor

jonathanlew, 2019-10-23:
### Description of the task
Spline(std::vector<Points> points, unsigned int order) constructor will take the points and interpolate with `order` degree polynomials with C(`order` - 1) continuity

### Acceptance criteria

- [ ] Make a Spline(std::vector<Points> points, unsigned int order) constructor
- [ ] Replace calls to Spline(std::vector<Points> points) with Spline(points, 1)
- [ ] Test it

### Blocked By
#942

MathewMacDougall, 2020-01-18:
Evan asked for a Spline.length() function to get the length of the spline

garethellis0, 2020-06-04:
Just as a note, this likely won't get done, we already have linear splines (essentially connected line segments) and I'm going to implement a `CubicBezierSpline`, but a generic spline class actually gets quite tricky in terms of specifying constraints in a user-friendly way.

akhilveeraghanta, 2020-07-24:
@garethellis0 can we close this ticket now, since we have the Bezier curve implementation?

garethellis0, 2020-07-29:
Yes, in retrospect this was overly ambitiously general.

closed 2020-07-29 by garethellis0