UBC-Thunderbots/Software #989: Switch From Clang To GCC To Fix Code Coverage Issues

garethellis0, 2019-10-25:
### Description of the task
In https://github.com/UBC-Thunderbots/Software/pull/985, we setup our toolchain to use `arm-none-eabi-gcc` for `arm` targets, and `clang` for `x86_64` targets. Unfortunately, bazel has issues getting proper code coverage from `clang`, and so we have a somewhat hacky script to manually convert the coverage outputs from `clang` into a format that CodeCov can understand. As such we would like to switch back to `gcc`, which bazel properly supports in terms of reading coverage (at least for some versions of gcc). We should use the newest version of gcc possible.

#### Relevant Bazel Issues 
https://github.com/bazelbuild/bazel/issues/9406
https://github.com/bazelbuild/bazel/issues/8217
https://github.com/bazelbuild/bazel/issues/7719

### Acceptance criteria

- [ ] Download `gcc` instead of `clang` via the `WORKSPACE` file the associated external build file
- [ ] Change the wrapper scripts in `cc_toolchain/wrappers` for clang to gcc
- [ ] Implement a `gcc` toolchain configuration in `cc_toolchain_config.bzl` and remove the `clang` one
- [ ] Change everything in `cc_toolchain/BUILD` from `clang` to `gcc`
- [ ] Remove the `convert_profdata_to_lcov` script, and make sure codecov still works (will probably need to add back the `-s bazel-testlogs` flag to the codecov command in `.travis.yml`)

### Blocked By

jonathanlew, 2020-02-08:
Will look into this after https://github.com/UBC-Thunderbots/Software/pull/1129 gets merged

MathewMacDougall, 2020-03-10:
Was this not resolved by #1184 ?

jonathanlew, 2020-03-11:
> 
> 
> Was this not resolved by #1184 ?

Yes

closed 2020-03-11 by jonathanlew