UBC-Thunderbots/Software #995: Ball history should be returned with the newest value first

MathewMacDougall, 2019-10-28:
### Description of the task

#922 adds nicer separation between a `BallState` at a single point in time and a `Ball` which stores history of multiple states. Currently this history is returned with the oldest data first, and the newest data last.

This is not consistent with how the data is stored internally to the ball (newest data at the "front" / first). Furthermore, most callers only need history starting from the latest data, so returning the data with the newest first is easier to work with. For example, a caller doesn't need data in the range [current_time - 1, current_time - 3].

The Ball history should changed to return the data with the newest data first and oldest data last. Other code that uses the history may need to be updated.

### Acceptance criteria

- [ ] Return state data with the newest first, and oldest last
- [ ] Update unit tests

### Blocked By

#809

MathewMacDougall, 2020-02-15:
Please also make sure robot history is returned newest first

kelvinkoon, 2020-02-22:
Ball history was found to already be returning by newest first.

closed 2020-02-22 by kelvinkoon