UBC-Thunderbots/Software #1008: Setup Protobuf Over Serial For STM32H7 Projects

garethellis0, 2019-11-02:
### Description of the task
We need a concrete way to send messages over serial to the robot in order to be able to control it.

### Acceptance criteria

- [ ] Setup a protobuf library in C++ that can send protobuf messages over serial (ex. `/dev/ttyACM0`)
- [ ] Setup a protobuf library that can run on the STM32H7 (ie. in embedded C)
- [ ] Succesfully send a message over protobuf to the STM32H7 and receive a protobuf from the STM32H7 in reply

### Blocked By

garethellis0, 2019-11-02:
This could be useful: https://github.com/protobuf-c/protobuf-c

akhilveeraghanta, 2019-11-30:
This is not feasible for our use case, closing in favor of the use of ethernet

closed 2019-11-30 by akhilveeraghanta