UBC-Thunderbots/Software #1010: New system for obstacle generation from motion constraints

jonathanlew, 2019-11-04:
### Description of the task
Currently obstacle generation is under navigator util. We can put that in the navigator/obstacle folder instead

Please consult @jonathanlew @garethellis0 and @MathewMacDougall on how this should look

### Acceptance criteria

- [ ] new system
- [ ] tests

### Blocked By

garethellis0, 2019-11-04:
Could we please be more specific with the "Acceptance Criteria" for this? These criteria should provide a clear set of things to need to be resolved, like "move obstacle generation functions from `util.{cpp,h}` to `obstacle_generation.{cpp,h}`. Please keep in mind that these tickets can be taken by new members with minimal context, and the more information we put here, the less time we spend explaining and the less time they have to spend asking.

jonathanlew, 2019-11-04:
Yup

jonathanlew, 2019-12-01:
Done by #1013

closed 2019-12-01 by jonathanlew