UBC-Thunderbots/Software #1022: Visualize obstacles with the visualizer

jonathanlew, 2019-11-09:
### Description of the task
Navigator passes obstacles to PathManager, who then passes it to the PathPlanner to path plan. Since path planning is done around obstacles, we want to see what the path planner is trying to avoid. We want to visualize the perimeter of all the obstacles in the visualizer. It's OK for all obstacles to have the same colour overlap on top of each other.

### Acceptance criteria

- [ ] Visualize all obstacles
- [ ] Visual inspection

### Blocked By

hardik-ddod, 2019-11-17:
@jonathanlew Do we have a dummy game scenario on which I could try running the current visualizer on my computer right now?

MathewMacDougall, 2019-11-18:
@hardik-ddod you can use `grsim for that. [The docs](https://github.com/UBC-Thunderbots/Software/blob/master/docs/getting-started.md#installing-grsim) explain how to install it. Then you can just run `grsim` from a terminal and it should open.

Then just build and run the `//software:full_system` target with the `--backend grsim` flag.

So the full command might look like `bazel run //software:full_system -- --backend grsim`, run from the `Software/src` folder.

hardik-ddod, 2019-11-18:
@MathewMacDougall TY papa leader!

jonathanlew, 2019-11-18:
yeah I synced up with hardik offline, but thanks

garethellis0, 2020-01-06:
@hardik-ddod - where's this at right now?

hardik-ddod, 2020-01-06:
@garethellis0 I am still trying to understand how everything works... not much work is done on this one...

garethellis0, 2020-01-06:
OK, could you please unassign yourself from this ticket until https://github.com/UBC-Thunderbots/Software/pull/1049 gets merged? Then Mathew M. would be the person to talk to about this.

PR #1352 (2020-05-28): Draw Geometry Shapes and Obstacles on the Visualizer 

## Please fill out the following before requesting review on this PR

### Description
* Change Obstacle to be a templated type on GEOM_TYPE, currently implemented with Polygon and Circle 
* Created abstractions for drawing geoms
* Integrated drawing geoms in drawing field, ball, robot 
* Draw Obstacles using an ObstacleVisitor

### Testing Done
Visual inspection  and unit tests for field

### Resolved Issues
Resolves #1022 

### Length Justification
I changed a function in Field so I ended up touching a ton of files 

### Review Checklist
closed 2020-05-28 by jonathanlew