UBC-Thunderbots/Software #1024: Document Convention for Use of Getters with or without get

jonathanlew, 2019-11-10:
### Description of the task
Some classes have getters prefixed by get, i.e. getName() and some don't, i.e. x(), y(), name(). 

### Acceptance criteria

- [ ] Document convention
- [ ] Refactor code to follow convention

### Blocked By

garethellis0, 2019-11-16:
I think that the advice should be to always prefix with `get`, in order to make functions consistent. @MathewMacDougall @akhilveeraghanta @jonathanlew - thoughts?

jonathanlew, 2019-11-16:
Have we scoped out what would change under that rule?

jonathanlew, 2019-11-16:
#915 is related to this issue

MathewMacDougall, 2019-11-17:
Ah wish I saw this before the meeting today we should have talked in person.

> I think that the advice should be to always prefix with get, in order to make functions consistent

This assumes that we are currently using `get` more often than not. Is that the case? Do we have an argument one way or the other that is not related to consistency?

Otherwise I agree about the classes that are really commonly used, but that's somewhat of an arbitrary cutoff. I don't think much would change under this rule because I think we currently use `get` for almost everything.

Which brings up another point, it seems we have chosen to use `get` even for boolean types sometimes, which are sometimes more commonly prefixed with `is`. Should we make a decision on that too here?

Overall I don't want to go too deep into this because it's not something I'm confident we can enforce, but we absolutely do want some level of consistency and sanity.

garethellis0, 2019-11-17:
I like the general concept of  using `get` as a prefix in most cases, `is` for boolean functions, and no prefix for _super_ commonly used functions. Without adding `clang-tidy` support this is tricky to support, but at least defining what's correct saves people from having this debate in the future!

garethellis0, 2019-11-17:
For anyone wondering, supporting clang-tidy is a ticket here: https://github.com/UBC-Thunderbots/Software/issues/839. If we think this is important enough I can prioritize it.

MathewMacDougall, 2019-11-18:
Can clang-tidy enforce naming conventions like this?

garethellis0, 2019-11-18:
Ah good point. Just checked and it doesn't look like it. Also this is a flexible requirement, so it would be pretty tricky to come up with a totally generic way of saying what is/is not valid.

MathewMacDougall, 2019-11-18:
> I like the general concept of using get as a prefix in most cases, is for boolean functions, and no prefix for super commonly used functions

I'm fine with this then, since it's most similar to what we already have and what people are used to. The only main discrepancy I think would be the use of `is`, but we can just address that as we see it. The action item here should be to document this convention if everyone agrees?

jonathanlew, 2019-11-18:
> I'm fine with this then, since it's most similar to what we already have and what people are used to. The only main discrepancy I think would be the use of `is`, but we can just address that as we see it. The action item here should be to document this convention if everyone agrees?

I'd just add that we have boolean-like values such as `BallCollisionType` enum and I think that using `get` for enums makes sense. To make our code _optimally_ readable (I'm not suggesting we should do this 😛 ), we'd have enum for every boolean and we could use get for all of them. But that's to say that if have a problematic boolean that uses `is`, then the solution is to make it an enum and use `get`

garethellis0, 2019-11-18:
That's a good point. I had to think about this a bit, but I think the dividing line for when something should be an enum vs a boolean is when one of:
a) There are potentially more then two logical states (even if we just have two in our code).
b) There is not a clear "true" value from a high level perspective.

This is definitely all very fuzzy and should be judged on a per-case basis though.

garethellis0, 2019-11-18:
I'm also not sure that replacing _all_ booleans with enums would make the code "optimally" readable :stuck_out_tongue:

PR #1054 (2019-11-20): Added Documentation On Getters And Setters To code-style-guide.md

## Please fill out the following before requesting review on this PR

### Description

### Testing Done

### Resolved Issues
resolves #1024

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function definitions (usually in the `.h` file) should have
closed 2019-11-20 by garethellis0