UBC-Thunderbots/Software #1032: Document Explicit Criteria For Each Difficulty Level For Issues

garethellis0, 2019-11-12:
### Description of the task
Currently, whether a task is a 3 or 7 is somewhat arbitrary and likely highly variant depending on who is classifying the ticket. We should explicitly document a set of criteria that can be used to help calibrate every so that they're at the same level.

### Acceptance criteria

- [ ] Note some explicit criteria for each difficulty level somewhere in the documentation
- [ ] Note some good examples for each level of difficulty in the documentation

### Blocked By

jonathanlew, 2020-08-16:
Updated the descriptions. @garethellis0 @akhilveeraghanta @MathewMacDougall please review
[image]

MathewMacDougall, 2020-08-17:
Seems sane to me. Do we want level `21` tickets to be broken up, or should that maybe be covered by the `planning` tag? Although generally I agree that what we have historically made level `21` have been sometimes very large tickets that could have been broken up if we had a better scope of the problem and solution.

jonathanlew, 2020-08-17:
A guideline I like to imagine existing (it doesn't and is hard to enforce) is one issue->one PR. So with that in mind, a level 21 ticket can not be done in one PR, and so should be broken down into smaller tickets that other people can do in one PR. A level 21 ticket usually can't be broken up immediately and should spawn smaller tickets once it is scoped out

MathewMacDougall, 2020-08-18:
>  level 21 ticket can not be done in one PR

Generally I agree, although it's possible for a level 21 ticket to exist due to lots of required context with a subsystem or other related knowledge (eg. optimization / gradient descent for passing). No strong opinions here since also hopefully most things that would be level 21 will be hashed out in a design doc going forwards.

jonathanlew, 2020-08-19:
It should be possible to break down level 21 tickets into at most level 13 tickets