UBC-Thunderbots/Software #1035: Add A

garethellis0, 2019-11-14:
### Description of the task

### Acceptance criteria

- [ ] Item A
- [ ] Item B
- [ ] Item C

### Blocked By

henryymliu, 2019-11-14:
We can use the `snap` version of valgrind (https://snapcraft.io/valgrind), it works on all distributions of Linux that support `snap`.

MathewMacDougall, 2019-11-14:
Can you freeze the version of a `snap` package? The issue might be that it could auto-update on someone's machine and break

henryymliu, 2019-11-14:
Not in ubuntu right now, but `sudo snap install valgrind --channel=stable/3.15.0 --classic` might work?
I also can't remember if Ubuntu has snap enabled by default, or if one has to `apt install snapd` first

garethellis0, 2019-11-15:
I think `sudo snap install valgrind --channel=stable --classic` should be fine, we really don't need to peg the version. I also had to run `sudo apt remove valgrind` to remove the version installed by apt-get (really it just depends which executable it hits first in the path).

PR #1105 (2019-12-26): Added Newer Version Of Valgrind To Software Setup Script

## Please fill out the following before requesting review on this PR

### Description

### Testing Done
Tested out the commands added.

### Resolved Issues
resolves #1035 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Function & Class comments**: All function definitions (usually in th
closed 2019-12-26 by garethellis0