UBC-Thunderbots/Software #1072: Investigate Using std::atomic Across The Codebase

garethellis0, 2019-11-28:
### Description of the task
[std::atomic](https://en.cppreference.com/w/cpp/atomic/atomic) provides a wrapper to make objects thread-safe. There are many places in the codebase, like in the `PassGenerator` and in several `MultithreadedObserver` classes, that we have locks for a single piece of data. We could likely just replace many of these locks with instances of `std::atomic`. 

The difficulty here is understanding when it is / is not appropriate to use `std::atomic`, as there are places where the locks are blocking for specific sections of code, not just single accesses.

### Acceptance criteria

- [ ] Find all locations in the repo that could benefit from this
- [ ] Either change all locations if it's not too much work, or file tickets to change all the locations

### Blocked By

garethellis0, 2019-12-24:
Closing this because it provides _much_ less benefit then many other tickets, and so is unlikely to be completed this year. @MathewMacDougall - please feel free to re-open if you disagree!

MathewMacDougall, 2019-12-24:
No need to re-open as you didn't close it :stuck_out_tongue: . I'd say leave it open for now since it should be a pretty quick fix in only a few places should we choose to address it, although I agree it's lower value.

jonathanlew, 2020-08-17:
std::atomic can only be used on trivially copyable functions and only guarantee sequentially consistency on multiple operations https://stackoverflow.com/questions/31978324/what-exactly-is-stdatomic, so they aren't appropriate for most of our code base where we have "transactions" that we want to happen atomically

MathewMacDougall, 2020-08-18:
Yea I'd be fine closing this since there's not a ton of value of trying to convert everything we can just to use `atomic`.

PR #1685 (2020-08-18): Scoped lock and pass generator lock cleanup

## Please fill out the following before requesting review on this PR

### Description
* Replace lock_guard with [scoped_lock](https://stackoverflow.com/questions/43019598/stdlock-guard-or-stdscoped-lock)
* Replaced fine-grained locks with one big lock in pass_generator
  * Should be easier to work with
  * `std::atomic` does not work because the types, e.g. `World`, `Point`, are not [trivially copyable](https://stackoverflow.com/questions/43019598/stdlock-guard-or-stdscoped-lock)

### Testing Done
No degradations in the running time of pass_generato [... cut]
closed 2020-08-18 by jonathanlew