UBC-Thunderbots/Software #1080: Document the Simulated Tests structure and assumptions

MathewMacDougall, 2019-12-03:
### Description of the task

We need to document the new structure and changes introduced in #1067 . This includes how the WorldStateValidator fits in to the structure and how the SimulatedTests fixture is set up (including timing between components).

### Acceptance criteria

- [ ] Add documentation

### Blocked By

MathewMacDougall, 2019-12-03:
@garethellis0 Thoughts on if this documentation should go in-code or in the markdown? A bit of both?

High-level structure for testing makes sense to go in markdown, but I'm not sure how much of the "assumptions made" stuff should go in markdown, since it's more tied to the implementation and _could_ change. Although the basic assumptions about timing / waiting on values are pretty generic so probably would be fine in markdown.

@jonathanlew @akhilveeraghanta FYI

garethellis0, 2019-12-03:
I think the cutoff should be that if you're talking about more then one class, that should go in the markdown, but not super strongly set on this.

PR #1103 (2020-01-06): Document the Simulated Testing framework and architecture

## Please fill out the following before requesting review on this PR

### Description

Documented the architecture and high-level design of our new simulated testing framework (features still WIP but design is locked down).

Relevant code can be found in `src/Software/simulated_tests` and `src/Software/backend`

### Testing Done

N/A

### Resolved Issues

resolves #1080 

### Length Justification

### Review Checklist
closed 2020-01-06 by MathewMacDougall