UBC-Thunderbots/Software #1085: Visualizer: Investigate timer error on shutdown

MathewMacDougall, 2019-12-05:
### Description of the task

Mathew TODO

```
QObject::~QObject: Timers cannot be stopped from another thread
FATAL CALL but logger is NOT initialized
CAUSE: SIGSEGV
Message: 
2019/12/05 12:59:31 929309

***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGSEGV(11)	PID: 22002

***** SIGNAL SIGSEGV(11)

*******	STACKDUMP *******
	stack dump [1]  /home/mathew/.cache/bazel/_bazel_mathew/84b6638ae5fbabc0fbab9d81e91fe5fc/execroot/__main__/bazel-out/k8-fastbuild/bin/software/full_system() [0xb35f52]
	stack dump [2]  /lib/x86_64-linux-gnu/libpthread.so.0+0x12890 [0x7f18d7a1e890]

	stack dump [3]  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 : QThreadStorageData::get() const+0x3b [0x7f18d63bb3eb]

	stack dump [4]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QOpenGLContext::currentContext()+0x2a [0x7f18d6b966aa]

	stack dump [5]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QOpenGLSharedResource::free()+0xb1 [0x7f18d6b98f31]

	stack dump [6]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QOpenGLTextureGlyphCache::clear()+0x1a [0x7f18d6ea77ba]

	stack dump [7]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QOpenGLTextureGlyphCache::~QOpenGLTextureGlyphCache()+0x17 [0x7f18d6ea7887]

	stack dump [8]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QOpenGLTextureGlyphCache::~QOpenGLTextureGlyphCache()+0x9 [0x7f18d6ea78b9]
	stack dump [9]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1c5a65 [0x7f18d6c21a65]

	stack dump [10]  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 : QHashData::free_helper(void (*)(QHashData::Node*))+0x49 [0x7f18d640d6b9]

	stack dump [11]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QFontEngine::~QFontEngine()+0x11c [0x7f18d6c1ac0c]
	stack dump [12]  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5+0xa7729 [0x7f18cc0f2729]

	stack dump [13]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 : QFontEngineMulti::~QFontEngineMulti()+0x55 [0x7f18d6c1ada5]
	stack dump [14]  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5+0xa2649 [0x7f18cc0ed649]
	stack dump [15]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1b189d [0x7f18d6c0d89d]
	stack dump [16]  /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1bad7f [0x7f18d6c16d7f]

	stack dump [17]  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 : QThreadStorageData::finish(void**)+0x2bd [0x7f18d63bb9cd]
	stack dump [18]  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0xab293 [0x7f18d63bc293]
	stack dump [19]  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0xab471 [0x7f18d63bc471]
	stack dump [20]  /lib/x86_64-linux-gnu/libpthread.so.0+0x6408 [0x7f18d7a12408]
	stack dump [21]  /lib/x86_64-linux-gnu/libpthread.so.0+0x781b [0x7f18d7a1381b]
	stack dump [22]  /lib/x86_64-linux-gnu/libc.so.6clone+0x3f [0x7f18d589788f]

exitWithDefaultSignalHandler:238. Exiting due to FATAL_SIGNAL, 11   

```

### Acceptance criteria

- [ ] Item A
- [ ] Item B
- [ ] Item C

### Blocked By

jonathanlew, 2020-01-25:
@MathewMacDougall Is this still a problem?

MathewMacDougall, 2020-01-26:
Unsure. My previous investigation suggested it's a dynamic parameter issue with the static singleton. THis is currently blocked on #960 so I can test again.

jonathanlew, 2020-07-30:
@akhilveeraghanta can you take this ticket?

MathewMacDougall, 2020-07-31:
Feel free to ping me whenever you do take a look at it. I don't have much more context to provide other than what I said in our last call, but should still be able to help.

MathewMacDougall, 2020-08-07:
@akhilveeraghanta de-assigning my tickets rn so I'll give this to you, but again feel free to loop me in when you tackle it

closed 2021-03-13 by akhilveeraghanta