UBC-Thunderbots/Software #1090: Visualizer: Make sure AI Control widgets are updated when parameters change

MathewMacDougall, 2019-12-11:
### Description of the task

Right now any widgets in the `AI Control` tab are not bound to their respective parameter callbacks, so they will not be immediately updated if something else changes the parameter value. These parameters should be connected, like the ones in the `Parameter` tab

### Acceptance criteria

- [ ] Connect the AI Control widgets to their parameter callbacks

### Blocked By

jonathanlew, 2020-09-10:
@akhilveeraghanta How does this fit into the new dynamic params?

MathewMacDougall, 2020-09-11:
No difference between old and new. It should just be a matter of connecting callback functions (which both the old and new versions have).

MathewMacDougall, 2020-09-11:
It's just that there may be params not connected to their respective DyanmicParam _at all_

jonathanlew, 2022-01-18:
new visualizer incoming

closed 2022-01-18 by jonathanlew