UBC-Thunderbots/Software #1091: Visualizer: Enable and disable drawing components

MathewMacDougall, 2019-12-11:
### Description of the task

Users should be able to toggle what is drawn in the Visualizer at runtime. For example, they may want to enable drawing navigation obstacles for debugging, but disable them later so the view isn't crowded.

These parameters will likely go in another tab for ease of access.

This can be achieved with DynamicParameters in the draw functions provided to the Visualizer. New parameters to control each component will need to be added. This should all be done after #960 

### Acceptance criteria

- [ ] Allow users to toggle what is drawn in the world view

### Blocked By

akhilveeraghanta, 2021-02-25:
Decided to not do because we are visualizing stuff in the jupyter notebook, will re-open if more layers are added

closed 2021-02-25 by akhilveeraghanta

MathewMacDougall, 2021-02-25:
Agree, this is probably unneeded complexity at this point.