UBC-Thunderbots/Software #1141: Refactor Ball Filter

MathewMacDougall, 2020-01-27:
### Description of the task

The Ball Filter still has a few large functions, and could use some minor re-structuring for readability.
* The publicly exposed function should be `estimateBallState` or `getEstimatedBallState` since this is more descriptive than `getFilteredData`
* The new `getEstimatedBallState` function can probably be broken down better into smaller functions without so many edge cases.

### Acceptance criteria

- [ ] Refactor ball filter
- [ ] Make sure tests still pass

### Blocked By

jonathanlew, 2020-07-30:
Are there additional areas in ball filter that can be refactored?

MathewMacDougall, 2020-07-31:
Nothing major comes to mind, mostly just breaking up a few large functions like I mentioned above. Also most of the functions should be made private.

kelvinkoon, 2020-08-16:
There is already a function in `ball_filter.cpp` called `estimateBallState`. I assume this ticket is about the `getFilteredData` function, renaming it to something like `getFilteredBall`, and breaking the function / cleaning it up.

MathewMacDougall, 2020-08-16:
It was generally about the whole thing. `estimateBallState` is the only function that really needs to be public, which on it's own is _slightly_ large and could probably be refactored along with other functions to be simpler. This definitely doesn't have to be a high-priority ticket.

closed 2020-09-14 by MathewMacDougall